moveit/moveit #2817: Does Moveit support calculating dynamics and inverse dynamics?

sm3304love, 2021-08-17:
Hello. I'm trying to use Moveit to build a cooperative robot. I have confirmed that Moveit supports kinematics and inverse kinematics, but I am not sure if it supports dynamics and inverse dynamics, so I ask a question. If both are not supported, is there any other way?

AndyZe, 2021-08-17:
It does not. Your best bet is to set up a Gazebo simulation. If you're lucky, somebody else will have already provided one. For example, the UR series of robots have Gazebo models.

v4hn, 2021-08-17:
not entirely true, but for sure nobody looked at our [dynamics_solver](https://github.com/ros-planning/moveit/tree/master/moveit_core/dynamics_solver) module in years.
So I would not expect it to work.