moveit/moveit #2818: Updating allowed collision matrix used for motion plans

aa-tom, 2021-08-17:
Hi,

I'm currently trying to generate a motion plan with collisions between a couple of links disabled. At the moment I'm using the following code:

```
    moveit::planning_interface::MoveGroupInterface move_group(req.move_group.data);
    planning_scene::PlanningScene scene(move_group.getRobotModel());
    std::string ee_link = move_group.getEndEffectorLink();
    for (std_msgs::String link : req.no_collision_links)
        scene.getAllowedCollisionMatrixNonConst().setEntry(ee_link, link.data, true);
    move_group.setPlanningPipelineId("pilz_industrial_motion_planner");
    move_group.setPlannerId("PTP");
    std::string ref_frame = move_group.getPlanningFrame();
    moveit_msgs::GetMotionSequence sequence;
     ....
```
`sequence` is then populated with a number of waypoints. The code fails to find a solution, presumably due to a collision as I've tested the code with the would-be colliding links removed from the scene and the trajectory is planned successfully. I'm guessing it might be because I'm creating a new scene rather than updating the one used by the planner? I'm not entirely sure how to do the latter, so any help would be much appreciated

felixvd, 2021-08-18:
Does this error occur with the changes in https://github.com/ros-planning/moveit/pull/2803 ?

simonschmeisser, 2021-09-26:
You seem to create a new ("empty") `PlanningScene` but it is not actually used. There can be many `PlanningScene`s but only one central one updated by `PlanningSceneMonitor` (PSM). I personally don't use the `MoveGroupInterface` but rather the `moveit_cpp` classes and underlying data structures so I might be off here.

There seems to be a "get_planning_scene" service and a "apply_planning_scene" service so you can apply changes to the "central" `PlanningScene` updated by the `PlanningSceneMonitor` in the `move_group` node. You basically want to take the allowed collision matrix (acm) from the result of the "get_planning_scene", modify it and create a new `moveit_msgs::PlanningScene` with your modified ACM and apply that via "apply_planning_scene". Note that this will apply "global" and "permanent" not just for path you're currently planning.

In the long run I recommend going more low level (moveit_cpp) where you can pass a `PlanningScene` directly to the planning pipeline. You would get that `PlanningScene` via `LockedPlanningSceneRO ps(your_psm)` instead of creating a new and "empty" one like in your sample code.