moveit/moveit #2823: "Unable to connect to move_group action server 'move_group'" when running on extern Rosmaster.

Kiwigamer, 2021-08-18:
I wanted to controll my RobotArm via Rviz, followed a tutorial and everything works fine.

In the end: I ran rviz on an external device with the rosmaster sitting on the robot itselfe.

For testing purposes i used the intern rosmaster and then to try it for "real" i got to connect to the rosmaster of the Robot "export ROS_MASTER_URI = http: //XXX.XXX.X.XX: 11311")".

Everything is the same except that when i run the demo.launch on the "real" rosmaster i got:
[ERROR] [1629309314.819565696]: Unable to connect to move_group action server 'move_group' within allotted time (30s)

Thats it. Nothing more shows up or happens and i am not able to find something that helps.
If you do have have any questions about my setup. I'll try to answer asap.

External Device (Virtual Machine):
Ubuntu 16.04.7 LTS
ROS PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

Robot (Raspberry Pi 4 B):
Ubuntu 16.04.7 LTS
ROS PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

Thanks,
Kiwi

felixvd, 2021-08-19:
This is almost certainly not a bug with MoveIt but a general networking problem with your ROS setup. Could you post on ROS Answers instead?

I don't have tutorials or links on hand, but from memory, you should make sure that:

- both machines in the network know each other (their hostnames appear in `etc/hosts` with the correct IP)
- both machines point to the same `ROS_MASTER_URI` (check environment variables like this with `echo $ROS_MASTER_URI`). It is better to set it to hostname than the IP (e.g. `export ROS_MASTER_URI=http://my-second-machine:11311`)
- both machines have `ROS_HOSTNAME` set to their own hostname
- no firewall/dropped packets, no other networks taking precedence over the ROS one, no time difference between the machines etc.

Often you will get the connection working from machine to the other but not the other way around. If you create a new question, please link it here and close this issue. If you solved your problem, don't forget to post your solution. Good luck

Kiwigamer, 2021-08-19:
Thanks @felixvd,
I'll go through your points and if it still doesn't work I'll switch to Ros Answers. (With that said i'll close the issue here)

closed 2021-08-19 by Kiwigamer

Kiwigamer, 2021-08-20:
So, I moved this question over to ROS Answers, because i haven't found a solution yet.

ROS Answers: https://answers.ros.org/question/384755/unable-to-connect-to-move_group-when-running-on-external-rosmaster/

Please post any further comments, ideas or questions there.

gavanderhoorn, 2021-08-20:
This was resolved on ROS Answers.

It turned out the OP had set `ROS_HOSTNAME` to `localhost`.

Kiwigamer, 2021-08-20:
Yeah ...