moveit/moveit #2831: git version same as last release?

v4hn, 2021-08-22:
why is [moveit's current version in the repo] (https://github.com/ros-planning/moveit/blob/1859ec2d248074c983367735f3034f3172de07b2/moveit/package.xml#L4)  1.1.5, when we that version was already released? didn't we agree some time ago to bump the versions right after each release?

v4hn, 2021-08-22:
[Here](https://github.com/ros-planning/moveit/issues/2036)'s the related issue of the original discussion.

I found this because #2037 introduced some API changes I wanted to account for in my code base.

rhaschke, 2021-08-22:
I vaguely remember... Looks like, we have followed this rule only _once_ in the past: after releasing version 1.0.4.
I guess the main reason that we didn't follow the rule in practice, is that the required procedure is much more complicated.
Maybe we should do it only if explicitly requested, like now? @tylerjw, what do you think?

rhaschke, 2021-08-22:
I added a corresponding PR: #2832.

v4hn, 2021-08-22:
Thanks! 

I wouldn't call one `sed` command + commit "much more complicated", but it's an additional step to take care of that is not enforced by the release tools... :confused: 
Also, the next person releasing should remember not to bump the version again.
Although that might actually be a valid practice and would result in odd numbers for releases and even ones for unreleased states.

Reopening the issue to finish the discussion on how to handle this in the future (or possibly how to help the release team remember it?)

rhaschke, 2021-08-22:
I think that's an item for the next maintainer meeting...