moveit/moveit #2837: Where is the 'sim_control_mode' parameter that is created by the setup_assistant used?

rickstaa, 2021-08-24:
### Description

While looking through the code-base of the [moveit_setup_assistant](https://github.com/ros-planning/moveit/tree/master/moveit_setup_assistant) I found the following lines are added to the `ros_controllers.yaml` configuration file:

```yaml
# Settings for ros_control hardware interface
hardware_interface:
  joints:
    - panda_joint1
    - panda_joint2
    - panda_joint3
    - panda_joint4
    - panda_joint5
    - panda_joint6
    - panda_joint7
    - panda_finger_joint1
  sim_control_mode: 1 # 0: position, 1: velocity
```

These files were added by @MohmadAyman in #894 but I could not find any information about where these parameters are used. I looked in both the [moveit](https://github.com/ros-planning/moveit/), [ros_control](https://github.com/ros-controls/ros_control) and [gazebo_ros_pkgs](https://github.com/ros-simulation/gazebo_ros_pkgs). Does anybody maybe know if these parameters are still required with the latest versions of these packages?

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.4
* Source or Binary build? Source
* If source, which branch? Master

v4hn, 2021-08-24:
The main patch of that GSoC project was merged prematurely because the mentor wanted to see it merged. :face_with_head_bandage: 

The parameter comes from [`ros_control_boilerplate`](https://github.com/PickNikRobotics/ros_control_boilerplate/blob/noetic-devel/src/sim_hw_interface.cpp#L53) as used in [`moveit_sim_controller` configs](https://github.com/PickNikRobotics/moveit_sim_controller/blob/melodic-devel/config/ur5_controllers.yaml#L13)

closed 2021-08-24 by v4hn

rickstaa, 2021-08-25:
@v4hn Ah, thanks a lot for pointing this out. I will keep it in the Noetic [panda_moveit_config](https://github.com/rickstaa/panda_moveit_config/tree/noetic-devel) then. 

If I understood your comment and issue #1937 correctly, this code is to be used with the [moveit_sim_controller](https://github.com/PickNikRobotics/moveit_sim_controller) when the `moveit_fake_controller_manager` is replaced by this package

v4hn, 2021-08-25:
> If I understood your comment and issue #1937 correctly, this code is to be used with the [moveit_sim_controller](https://github.com/PickNikRobotics/moveit_sim_controller) when the `moveit_fake_controller_manager` is replaced by this package

That's unlikely to ever happen now because MoveIt2 relies on ros_control to provide the simulation, but yes.
You should actually be able to get `moveit_sim_controller` to run already. But I didn't use it in quite some time, so I don't know.