moveit/moveit #2860: Moveit Servo Follow Path

nihalar, 2021-08-31:
Hi All,

I have a list of end effector waypoints, I want the end effector to follow the waypoint path one after the other. I'm using pose tracking API for the same.  When I publish multiple points to the moveit servo, the robot directly goes to the last point and ignores intermediate points. How would I make the robot follow all the points in a path? 

Thank You,

AndyZe, 2021-08-31:
First, did you see the example in moveit_servo/src/cpp_interface_example/pose_tracking_example.cpp? It doesn't have multiple waypoints but it's a good first start.

Moving between waypoints is going to be a little tricky because of the multithreaded nature of pose tracker.

Let's look at the function signature of moveToPose():

```
PoseTrackingStatusCode PoseTracking::moveToPose(const Eigen::Vector3d& positional_tolerance,
                                                const double angular_tolerance, const double target_pose_timeout)
```
                                                
So it returns PoseTrackingStatusCode::SUCCESS when it reaches the waypoint within the tolerance you provided. You need to communicate that to the thread that manages your waypoints. Maybe something like this:

```
  std::atomic<bool> waypoint_reached{false};

  ros::Rate rate(50);
  for (size_t waypoint_id = 0; i < num_waypoints; ++waypoint_id)
  {
    // Get the next target pose here

   // Reset waypoint_reached flag
   waypoint_reached = false;

    // Run the pose tracking in a new thread
    std::thread move_to_pose_thread(
      [&tracker, &lin_tol, &rot_tol, &waypoint_reached]
      {
        auto tracker_status = tracker.moveToPose(lin_tol, rot_tol, 0.1 /* target pose timeout */);
        if (tracker_status == PoseTrackingStatusCode::SUCCESS)
          waypoint_reached = true;
      });

    target_pose_pub.publish(target_pose);
    while (!waypoint_reached)
    {
      rate.sleep();
     }
  }
```

AndyZe, 2021-08-31:
If your waypoints are static, i.e. known in advance, you could use the Pilz industrial motion planner to plan between them and blend the path. That's probably a better option if the waypoints are static.

nihalar, 2021-08-31:
> First, did you see the example in moveit_servo/src/cpp_interface_example/pose_tracking_example.cpp? It doesn't have multiple waypoints but it's a good first start.
>
> Moving between waypoints is going [...]

Thanks for the reply.  I did look at the examples. As the error gets closer to the tolerance, the velocity would become smaller. So, Wouldn't the robot stop (zero or next to zero velocity) between the waypoints?

Thanks :)

closed 2021-10-04 by nihalar