moveit/moveit #2861: OMPL constrained planning path tolerance

gautz, 2021-08-31:
While working on our welding use-case I could observe that the position constraints are not fulfilled with high accuracy. The line is being followed with a tolerance of about 1cm, and up to 2cm. See:
https://user-images.githubusercontent.com/64833674/131546004-627b00eb-ee1d-43e0-9967-5fcbb4a37a9b.mp4
https://user-images.githubusercontent.com/64833674/131546016-4ede0934-a07e-4d51-a3d8-1b4b9b2a5d92.mp4
[image]

Did you observe this already ? I think it is a slight deviation that may not be visible in the [tutorial](https://github.com/ros-planning/moveit_tutorials/pull/651). 

Is there some numerical parameter in the MoveIt-OMPL interface that could lead to such an error? Like e.g. a path_tolerance with default value 0.01 or so? @JeroenDM @mamoll

gautz, 2021-08-31:
Note that the above errors are obtained with a ``goal_orientation_tolerance: 0.5``
When I set ````goal_orientation_tolerance: 0.5`` the position error comes down to some mm:

https://user-images.githubusercontent.com/64833674/131552675-23e1b92b-b5b6-4a73-a4d4-7c00525f59cf.mp4

My feeling is that the second case is easier to optimize as the motion in joint space is simpler.

v4hn, 2021-08-31:
> Note that the above errors are obtained with a goal_orientation_tolerance: 0.5
> When I set ````goal_orientation_tolerance: 0.5`` the position error comes down to some mm

What? I guess you didn't mean to write the same line twice here?

It's hard to see anything in your rviz videos and plots of the eef trajectory would probably be more useful.

Sorry I can't help though, I did not use these interfaces myself yet :(

gautz, 2021-09-01:
Yes I meant 0.005 in the second case, thanks. I'll generate some plots.

JeroenDM, 2021-09-28:
The main tolerance value used for constraint satisfaction value is `1e-4 m`, the default value from OMPL, but the whole situation with the constraint threshold is a bit of a mess because of the API hack, as explained [here](https://github.com/JeroenDM/moveit/blob/bb594a81c52f8c9258ffac752bf2fe3bda40c3dd/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/ompl_constraints.h#L301). (But I'm must admit I find it difficult to parse that comment, and I wrote it...)

I'm also curious for numerical values or plots if you manage to get these.

**Some extra details**
The tolerance value is used in OMPL [here](https://github.com/ompl/ompl/blob/9c3a20faaddfcd7f58fce235495f043ebee3e735/src/ompl/base/src/Constraint.cpp#L104), where `f` is the error vector, aka, the deviation from the (position constraints only?) in this case.
```C++
const double squaredTolerance = tolerance_ * tolerance_;
while ((norm = f.squaredNorm()) > squaredTolerance && iter++ < maxIterations_)
```

Assuming you are using equality constraints, the calculation of this error vector `f` is implemented [here](https://github.com/JeroenDM/moveit/blob/bb594a81c52f8c9258ffac752bf2fe3bda40c3dd/moveit_planners/ompl/ompl_interface/src/detail/ompl_constraints.cpp#L243) with `f = pos_error` below.
```c++
// Position error calculated in the robot base link frame.
Eigen::Vector3d pos_error = forwardKinematics(joint_values).translation() - target_position_;
// Position error expessed in the nominal goal frame along the path.
pos_error = target_orientation_.matrix().transpose() * pos_error;
```
I would not expect `goal_orientation_tolerance` to infuence the position error, but this last piece of code might explain why it does. Although I have to think about it a bit longer to understand it. Hope this helps already for now.