moveit/moveit #2866: Plan/Control a frame which is not a robot link

gautz, 2021-09-07:
Let's suppose I publish at runtime a TF frame ``task_control_frame``, child of the robot tcp link frame. Can I plan a motion which moves this frame (rather than a robot link frame) towards the planning goal ? Similar to what is done in MTC with setIKFrame() https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stages/move_to.cpp#L210-L247

See this question https://answers.ros.org/question/385951/task-control-frame-for-moveit-mtc-move-group-and-servo/

rhaschke, 2021-09-07:
MoveIt / MTC cannot know that this transform is static. MoveIt's PlanningScene has its [own set of fixed transforms for this purpose](https://docs.ros.org/en/noetic/api/moveit_core/html/cpp/classplanning__scene_1_1PlanningScene.html#aa8aa5cd7aab5f814bdf6cf30fd501e9e). Just set the transform via this mechanism and it should work.

v4hn, 2021-09-07:
I just gave a similar answer on answers.ros.org.
Closing. If you find either method not working, please file a concrete issue about it.

closed 2021-09-07 by v4hn

felixvd, 2021-09-09:
I find this feature quite hard to discover and that it has room for improvement, so a tutorial or example might be a good idea as well.

gautz, 2021-09-09:
I tried to implement a ``planning_scene_monitor`` but did not manage to read/write the ``fixed_frame_transforms``.
@rhaschke can you be more specific about the most efficient way to publish the fixed transform?
Here is what I tried. I added in main:
```
    planning_scene_monitor::PlanningSceneMonitorPtr planning_scene_monitor;
    planning_scene_monitor = std::make_shared<planning_scene_monitor::PlanningSceneMonitor>("robot_description");
    if (!planning_scene_monitor->getPlanningScene())
    {
        ROS_ERROR_STREAM("Error in setting up the PlanningSceneMonitor.");
        exit(EXIT_FAILURE);
    }
    ROS_INFO_STREAM("Planning frame of monitored scene: " << planning_scene_monitor->getPlanningScene()->getPlanningFrame());
    planning_scene_monitor->startSceneMonitor("/move_group/monitored_planning_scene");
    planning_scene_monitor->startWorldGeometryMonitor(
        planning_scene_monitor::PlanningSceneMonitor::DEFAULT_COLLISION_OBJECT_TOPIC,
        planning_scene_monitor::PlanningSceneMonitor::DEFAULT_PLANNING_SCENE_WORLD_TOPIC,
        false /* skip octomap monitor */);
    planning_scene_monitor->startStateMonitor();
    planning_scene_monitor->providePlanningSceneService();
```
and in my method:
```
    geometry_msgs::TransformStamped task_control_fixed_frame_transform;
    task_control_fixed_frame_transform.header.frame_id = "tcp_welding_gun_link";
    task_control_fixed_frame_transform.header.stamp = ros::Time::now();
    task_control_fixed_frame_transform.child_frame_id = "task_control_frame";
    task_control_fixed_frame_transform.transform = task_transform_inv;
    planning_scene_monitor_->requestPlanningSceneState();
    planning_scene_monitor::LockedPlanningSceneRW ps(planning_scene_monitor_);
    planning_scene_monitor_->updateFrameTransforms();
    moveit_msgs::PlanningScene scene_msg;
    ps->getPlanningSceneMsg(scene_msg);
    scene_msg.fixed_frame_transforms.push_back(task_control_fixed_frame_transform);
    bool ok;
    planning_scene_monitor_->updateFrameTransforms();
    ok = planning_scene_monitor_->newPlanningSceneMessage(scene_msg);
    if (!ok)
        ROS_INFO_STREAM("Error in sending planning scene message.");
    ros::Duration(0.5).sleep();
    planning_scene_monitor_->triggerSceneUpdateEvent(
        (planning_scene_monitor::PlanningSceneMonitor::SceneUpdateType)(planning_scene_monitor::PlanningSceneMonitor::UPDATE_TRANSFORMS));
    ROS_INFO_STREAM("Transforms updated in scene.");
    planning_scene_monitor_->unlockSceneRead();
    planning_scene_monitor_->unlockSceneWrite();
    ROS_INFO_STREAM("Planning scene unlocked.");
    planning_scene_monitor_->stopSceneMonitor();
```

@felixvd I am thinking of rather using the subframes_tutorial as a basis. But is it possible to use subframes as ``robot_link_command_frame`` in ``moveit_servo`` ?

felixvd, 2021-09-10:
I haven't looked at how `moveit_servo` does things (direct IK?), but if it uses the `PlanningScene` then it should work, since it resolves the subframes.

AndyZe, 2021-09-10:
Servo uses RobotState `getGlobalLinkTransform()`. So I think it will only work for links that are attached to the robot. It won't work for a tf frame published from somewhere else.

https://github.com/ros-planning/moveit/blob/ffe4697696147f0e4db116a19f70cde7c5f5d074/moveit_ros/moveit_servo/src/servo_calcs.cpp#L196

@felixvd do you know if `getGlobalLinkTransform()` would work for a subframe? I guess that's the crux of the question.

felixvd, 2021-09-12:
[getGlobalLinkTransform](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L1387) passes the call directly to the `robot_model`, which does not know subframes, so no. But if you used [getFrameTransform](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L1751) instead, it should resolve (see [here](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/src/robot_state.cpp#L1080), then a few lines down [here](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/src/robot_state.cpp#L1125)). The result's reference frame should be the same. Maybe you can try that out @gautz .