moveit/moveit #2867: Default distance metric for ompl_interface::JointModelStateSpace

vinnamkim, 2021-09-08:
MoveIt uses L1 distance metric for `ompl_interface::JointModelStateSpace`. But, it seems somewhat less common than using Euclidean (L2) distance metric. There seems some cases that could be undesirable if we keep L1 distance metric.

1. Interpolation
 - `ompl_interface::JointModelStateSpace` interpolation follows a linear interpolation. If we have two states `s1` and `s2`, and an interpolated state `s3 := interpolate(s1, s2, 0.5)`, L1 distance metric makes `distance(s1, s3) / distance(s1, s2) != 0.5`.
2. `checkMotion(s1, s2)`
 - MoveIt's OMPL planner's default behavior to check feasibility of a motion between two states `checkMotion(s1, s2)` is `ompl::base::DiscreteMotionValidator` https://github.com/ompl/ompl/blob/9c3a20faaddfcd7f58fce235495f043ebee3e735/src/ompl/base/src/DiscreteMotionValidator.cpp#L93-L103
 - But, L1 distance metric makes under-segmentation for the large change in a single dimension than small changes with multiple dimensions.
 - Therefore, The large change in a single dimension will result in fewer collision checks, and this is getting worse when the number of joints are increasing.
3. Connect two states in the planner's tree
 - It is same with discussed in 2. The planner tends to connect two states only move in a fraction of the total dimension.

Additionally, there can exist undesirable cases in [`ompl::geometric::PathSimplifier`](https://github.com/ompl/ompl/blob/9c3a20faaddfcd7f58fce235495f043ebee3e735/src/ompl/geometric/src/PathSimplifier.cpp) (It uses `distance()` and `interpolate()` functions many, but I didn't look closely).

v4hn, 2021-09-08:
> MoveIt uses L1 distance metric for `ompl_interface::JointModelStateSpace`. But, it seems somewhat less common than using Euclidean (L2) distance metric.

I'm not sure why you deem Euclidean distance the better choice for Joint spaces, though I agree that it would be great to support other/possibly even custom distance functions. Some additional discussion can be found [here](https://github.com/ros-planning/moveit/pull/1758).

Please don't just give such hard-to-verify statements, but provide concrete tests/examples that illustrate the issue.

vinnamkim, 2021-09-08:
My main claim was 1. but after looking again, I realized that I made a mistake. 
It is actually `distance(s1, s3) / distance(s1, s2) == 0.5.` in L1 metric.
Close this issue.

closed 2021-09-08 by vinnamkim