moveit/moveit #2874: Moveit Build Failed

kashifzr, 2021-09-14:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source Build

### Steps to reproduce
Colcon Build 

### Expected behaviour
successful build 

### Actual behaviour
Failing Build 

### Backtrace or Console output

<pre><font color="#CC0000">...............................................................................</font>
<font color="#CC0000"><b>Failed</b></font>    <font color="#CC0000"> &lt;&lt;</font> <font color="#06989A">chomp_motion_planner</font>:<font color="#3465A4">make                                      </font>[ <font color="#C4A000">Exited with code </font><font color="#C4A000"><b>2</b></font> ]
<font color="#CC0000"><b>Failed</b></font>    <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">chomp_motion_planner                                          </font> [ <font color="#C4A000">12.8 seconds</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit                                                        </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_planners                                               </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_plugins                                                </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_commander                                              </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_chomp_optimizer_adapter                                </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros                                                    </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_fake_controller_manager                                </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_kinematics                                             </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_planners_ompl                                          </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros_move_group                                         </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros_manipulation                                       </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros_perception                                         </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros_robot_interaction                                  </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros_warehouse                                          </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros_benchmarks                                         </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros_planning_interface                                 </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_planners_chomp                                         </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros_visualization                                      </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_runtime                                                </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_servo                                                  </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_setup_assistant                                        </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">panda_moveit_config                                           </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">pilz_industrial_motion_planner_testutils                      </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">pilz_industrial_motion_planner                                </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_visual_tools                                           </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#CC0000"><b>Abandoned</b></font> <font color="#CC0000">&lt;&lt;&lt;</font> <font color="#06989A">moveit_tutorials                                              </font> [ <font color="#C4A000">Unrelated job failed</font> ]
<font color="#C4A000"><b>_______________________________________________________________________________</b></font>
<font color="#C4A000"><b>Warnings</b></font>  <font color="#C4A000"> &lt;&lt;</font> <font color="#06989A">moveit_ros_planning</font>:<font color="#3465A4">make</font> <font color="#C4A000"><b>/home/abc/ws_moveit/logs/moveit_ros_planning/build.make.000.log</b></font>
/usr/bin/ld: warning: libconsole_bridge.so.0.4, needed by /opt/ros/noetic/lib/libclass_loader.so, may conflict with libconsole_bridge.so.1.0
<font color="#2E3436"><b>cd /home/abc/ws_moveit/build/moveit_ros_planning; catkin build --get-env moveit_ros_planning | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -</b></font>

<font color="#C4A000">...............................................................................</font>
<font color="#2E3436"><b>Finished</b></font>  <font color="#4E9A06">&lt;&lt;&lt;</font> <font color="#06989A">moveit_ros_planning                                           </font> [ <font color="#C4A000">20.3 seconds</font> ]
[build] <font color="#C4A000"><i><b>Summary:</b></i></font> <i><b>19</b></i> <i>of </i><i><b>46</b></i> <i>packages succeeded.</i>                                  
[build]   <font color="#2E3436"><i><b>Ignored:   None.</b></i></font>                                                     
[build]   <font color="#C4A000"><i><b>Warnings:</b></i></font>  <i><b>20</b></i> <i>packages succeeded with warnings.</i>                      
[build]   <font color="#CC0000"><i><b>Abandoned:</b></i></font> <i><b>26</b></i> <i>packages were abandoned.</i>                               
[build]   <font color="#CC0000"><i><b>Failed:</b></i></font>    <i><b>1</b></i> <i>packages failed.</i>                                        
[build] <i><b>Runtime:</b></i> <i>1 minute and 44.7 seconds total.</i>                              
[build] <i><b>Note:</b></i> <i>Workspace packages have changed, please re-source setup files to use them.</i>
</pre>

simonschmeisser, 2021-09-18:
@kashifzr did you figure out the issues with your setup?

kashifzr, 2021-09-19:
@simonschmeisser  In case of Ubuntu 20.04 I am unable to build Moveit from Source.However Binary Installation Works

simonschmeisser, 2021-09-20:
Please reopen if you have more information or need further help, currently it's hard to guess what's wrong

closed 2021-09-20 by simonschmeisser