moveit/moveit #2875: planning_scene_interface.getObjectPoses() should return CollisionObject.pose ?

gautz, 2021-09-15:
### Description

We were previously using following to retrieve the pose of our workpiece: `std::map<std::string, geometry_msgs::Pose> object_pose = planning_scene_interface.getObjectPoses(std::vector<std::string> ids);`
Since last MoveIt upgrade, the pose is always at the origin.

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Master

### Steps to reproduce
`planning_scene_interface.getObjectPoses(std::vector<std::string> ids);` to retrieve the pose of a collision object.
https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/planning_scene_interface/src/planning_scene_interface.cpp#L159-L164

### Expected behaviour
`planning_scene_interface.getObjectPoses()` should return: 
- `CollisionObject.pose`
- or a composition of `CollisionObject.pose` and `CollisionObject.primitive_poses`/`CollisionObject.mesh_poses`

### Actual behaviour
The pose returned is `CollisionObject.mesh_poses` or `CollisionObject.primitive_poses` which is for us always at the origin.

gautz, 2021-09-15:
I think this is related to https://github.com/ros-planning/moveit/pull/2037/

felixvd, 2021-09-15:
Right, I overlooked this function. If you file a PR I'm happy to merge it.

closed 2021-09-18 by simonschmeisser