moveit/moveit #2879: PILZ Planning with attached collision object subframe

gautz, 2021-09-20:
### Description

Trying to use an attached collision object subframe as an end-effector link for planning. It works as expected with OMPL, but with Pilz it seems like the transformation of the subframe is just applied w.r.t the world frame and not w.r.t the attached object pose.

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Master

### Steps to reproduce
- add `<arg name="planning_adapters" value="default_planner_request_adapters/ResolveConstraintFrames" />` to both pilz and ompl `planning_pipeline.launch.xml`
- set `move_group.setEndEffectorLink("cylinder/tip")` as in the subframe tutorial, e.g. shifted from TCP frame along `z=0.1`.
- Move to a given target with OMPL and compare with PILZ

### Expected behaviour
- The robot moves to the same target with same IK frame with PILZ and OMPL

### Actual behaviour
- The robot moves to the target with a different IK frame, e.g. OMPL moves to expected pose, but PILZ moves ``0.2`` below desired pose.

gautz, 2021-09-21:
I'll further try out or provide a test as soon as the subframe tutorial is running again. See https://github.com/ros-planning/moveit_tutorials/pull/520

robertjbush, 2021-09-22:
We have replicated this (*strange*) issue. But as of yet don't have ideas on the root cause or solution.

aa-tom, 2021-09-28:
Video of (what I believe is) the same bug in action:

https://user-images.githubusercontent.com/86778266/135055432-d04fd6b1-5f9e-4ba4-838f-64af3624f8ea.mp4

I'm planning to start looking into this today. If anyone knows which files would be relevant, that'd be super helpful, otherwise I'll do a search for code that looks subframes related

aa-tom, 2021-09-28:
I'm wondering if [this](https://github.com/ros-planning/moveit/blob/d1e1c3d27ab33978fdcc5d7f9494afaa76496adc/moveit_planners/pilz_industrial_motion_planner/src/trajectory_functions.cpp#L81) line of code could be the issue. Maybe we need to call `rstate.setToIKSolverFrame()` first to ensure that the goal pose is in the sover's co-ordinate frame?

aa-tom, 2021-09-28:
Interestingly, I added some log statements to `computePoseIK()` and to the code I'm using to generate the motion plan and, while the plan contains the subframe as the link name, Pilz seems to be the end effector link of the move group instead:

```
[ WARN] [1632828833.658387895] [/fake/tool_routine]: Request link name: ee4/chuck
[ WARN] [1632828862.855474672] [/fake/move_group]: Link name used for IK solving: axia80_mate
```

aa-tom, 2021-09-28:
It also looks like the robot state built on [this](https://github.com/ros-planning/moveit/blob/d1e1c3d27ab33978fdcc5d7f9494afaa76496adc/moveit_planners/pilz_industrial_motion_planner/src/trajectory_functions.cpp#L70) line doesn't see the attached collision objects, which might be part of the reason why this isn't working. I've switched it out for a call to `scene->getCurrentState()`, which allows the state to track the objects, but this hasn't fixed the original issue. Perhaps a step in the right direction though.

aa-tom, 2021-09-29:
After a bit more investigation, I've found out that we're using the default_planner_request_adapters/ResolveConstraintFrames plugin for our requests, which I believe will be transforming any goal constraints defined on subframes into equivalent goal points defined for the end effector link of the move group, which would explain why the goal links are appearing as axia80_mate on the pilz side, and why the planner doesn't need to know about the ACOs. Does mean we're back to square one though :sweat_smile:

commit e30c6fa8 (2021-11-06): Fix orientation of subframe offset in Pilz planners (#2890)

Fix #2879 by reorienting the subframe offset applied to a goal pose in the PTP planner,
so that it is applied in the correct direction.
Also add the same offset computation to LIN and CIRC generators,
thus allowing them to work with subframes.
closed 2021-11-06 by rhaschke