moveit/moveit #2881: macos-latest robostack CI job is broken

v4hn, 2021-09-20:
As I pointed out in https://github.com/ros-planning/moveit/pull/2876#issuecomment-919996654 , the macos-latest CI job fails for a week now with this message:

    ninja: error: '/Users/runner/micromamba/envs/test/lib/liborocos-kdl.1.5.0.dylib', needed by 'devel/lib/python3.8/site-packages/pymoveit_core.cpython-38-darwin.so', missing and no known rule to make it

@wolfv @Tobias-Fischer Does the robostack get notified by failures in our CI? If not, could you please create a patch to either send mails on failure to a mailing list the responsible devs are reading or at least add a comment to the workflow file saying whom we should ping to resolve issues in the future?

Tobias-Fischer, 2021-09-20:
Hi @v4hn, thanks for pinging us! We'll look into this. Currently we don't get notified by failures - generally pinging @wolfv, @traversaro and myself would be great.

@traversaro - on gitter you suggested to patch this:
https://github.com/orocos/orocos_kinematics_dynamics/blob/db25b7e480e068df068232064f2443b8d52a83c7/orocos_kdl/src/CMakeLists.txt#L60-L65

I am not quite sure what needs doing there - if you let me know what the exact patch should look like, I can take care of opening a PR in the orocos-kdl feedstock to fix this issue.

traversaro, 2021-09-20:
> @traversaro - on gitter you suggested to patch this:
> https://github.com/orocos/orocos_kinematics_dynamics/blob/db25b7e480e068df068232064f2443b8d52a83c7/orocos_kdl/src/CMakeLists.txt#L60-L65
>
> I am [...]

PR: https://github.com/conda-forge/orocos-kdl-feedstock/pull/16 .

v4hn, 2021-09-21:
> Currently we don't get notified by failures

Is it possible to modify the CI job specification to send mails to you on failure?
Skimming online documentation for GHA I can't find a way to do that, but I'm probably just not capable enough? @JafarAbdi @rhaschke

wolfv, 2021-09-21:
For the time being we could also pin the specific version of orocos-kdl that used to work, right @traversaro ?

Tobias-Fischer, 2021-09-21:
Let's check: https://github.com/ros-planning/moveit/pull/2883

closed 2021-09-22 by rhaschke

v4hn, 2021-09-22:
The issue about sending mails to the three of them when jobs break remains unsolved.

rhaschke, 2021-09-22:
I couldn't find an easy way to send email notifications on status changes either. There doesn't seem to be a built-in functionality for this. However, there are dozens of notify actions available: https://github.com/marketplace?type=actions&query=notify+
One "just" needs to find the right one, which filters by changes of the build status.

v4hn, 2021-09-22:
> However, there are dozens of notify actions available: https://github.com/marketplace?type=actions&query=notify+
One "just" needs to find the right one, which filters by changes of the build status.

It would be very much appreciated if someone from the robostack team could create a PR setting up such a notification (whatever channel suits you) for their CI job. I would propose to have it trigger for the main branches (not for PRs) only, to avoid unrelated failures in someones PR.

wolfv, 2021-10-20:
yeah, there are some network issues apparently (we've also hit that in another project). 

Working on fixing this in https://github.com/mamba-org/provision-with-micromamba/pull/27

We will also add another mirror for obtaining micromamba soon. Sorry

rhaschke, 2021-10-21:
I disabled this workflow for now. @wolfv, please tell me when this is fixed and can be re-enabled. 
And please configure email notifications for this workflow as requested by this issue.

PR #2883 (2021-09-22): Temporarily pin orocos-kdl to fix cross-platform CI
### Description

Fix #2881 by pinning orocos-kdl to a working version.