moveit/moveit #2886: 1.1.6 Release

aa-tom, 2021-09-28:
It seems like there's a good amount of features & bugfixes on master since the release of 1.1.5, so I was wondering if it'd be possible to release a new version? If so, what would the process be? Can anyone pick up the task?

Thanks!

rhaschke, 2021-11-06:
I have put way more effort into this release than expected / intended. I didn't merge any item from your list, Michael - except the first one (which is still waiting for some unittests though), but fixed several other issues #2940, #2932, #2945, #2946.
- moveit_resources: https://github.com/ros/rosdistro/pull/31146
- moveit: https://github.com/ros/rosdistro/pull/31147

Altogether, this release comprises 290 commits!
We just have had a [release sync on Nov, 3rd](https://discourse.ros.org/t/preparing-for-noetic-sync-2021-11-03/22948). So, we will have enough soaking time for this monster release :wink:

closed 2021-11-06 by rhaschke

v4hn, 2021-11-07:
> but fixed several other issues

some of them where not even mentioned recently before you wrote patches for them and merged them without external review...
I'm grateful for all the work you put in here and horray for the release, but I would not have claimed any of your mentioned prs are release blocking and thus should be merged without review. :pensive:

rhaschke, 2021-11-07:
I merged those PRs accidentally when trying to push [my merge branch](https://github.com/ubi-agni/moveit/commit/80a64560a7c526633719b77618d78785f44c5c01) for testing to my fork. However, VS code didn't consider my push remote (which I configured to prevent exactly this), but pushed to the tracked remote. :disappointed: Sorry.
If you have any comments on the commits, I'm happy to get feedback in hindsight as well! Most of these commits address issues from the lists in https://github.com/ros-planning/panda_moveit_config/issues/89 and https://github.com/ros-planning/moveit/issues/2741.