moveit/moveit #2889: Document linear nature of Bullet CCD integration

werner291, 2021-09-29:
Hello,

I recently noticed that the CCD parts of the `collision_detection_bullet` seem to use a convex hull approximation. For reference: 
 - https://github.com/ros-planning/moveit/blob/049be73b8343d67ffde400fed9ed4c387c6ea9e5/moveit_core/collision_detection_bullet/include/moveit/collision_detection_bullet/bullet_integration/bullet_utils.h#L294
 - https://github.com/ros-planning/moveit/blob/049be73b8343d67ffde400fed9ed4c387c6ea9e5/moveit_core/collision_detection_bullet/include/moveit/collision_detection_bullet/bullet_integration/bullet_utils.h#L312

This can be a rather crude approximation when the motion includes a lot of rotations. For instance, consider a robotic arm that lays nearly flat on the ground, then swings almost 180 degrees through the air to lay almost flat in the opposite direction. In this case, the convex hull approximation won't cover the vast majority of the swept volume.

I understand that CCD is quite a complicated topic, so this method might be reasonable depending on circumstances, but I think it would make sense to at least document the fact that this approximation is used such that a user can act accordingly.

simonschmeisser, 2021-09-29:
Sounds like maybe there should be some maximum distance and CCD only be used in between interpolated points of that maximum distance?

werner291, 2021-10-06:
So, what I ended up doing was something like this:

```
double BulletContinuousMotionValidator::estimateMaximumRotation(const moveit::core::RobotStatePtr &st1,
                                                                const moveit::core::RobotStatePtr &st2) {// Upper bound on the rotation of any part of the robot.
    assert(st1->getRobotModel() == st2->getRobotModel());

    double max_angle = 0.0;
    for (const auto &jm: st1->getRobotModel()->getJointModels()) {
        switch (jm->getType()) {
            case moveit::core::JointModel::UNKNOWN:
                ROS_ERROR("Cannot compute maximum rotation with unknown joint type.");
                break;
            case moveit::core::JointModel::PRISMATIC:
            case moveit::core::JointModel::FIXED:
                // These do not affect rotation
                break;

            case moveit::core::JointModel::PLANAR: {
                const double *variables1 = st1->getJointPositions(jm);
                const double *variables2 = st2->getJointPositions(jm);
                max_angle += std::abs(variables1[2] - variables2[2]);
            }
                break;

            case moveit::core::JointModel::FLOATING: {

                const double *variables1 = st1->getJointPositions(jm);
                Eigen::Quaterniond rot1(variables1[6], variables1[3], variables1[4], variables1[5]);
                const double *variables2 = st2->getJointPositions(jm);
                Eigen::Quaterniond rot2(variables2[6], variables2[3], variables2[4], variables2[5]);

                max_angle += rot1.angularDistance(rot2);
            }
            break;

            case moveit::core::JointModel::REVOLUTE: {
                const double *variables1 = st1->getJointPositions(jm);
                const double *variables2 = st2->getJointPositions(jm);
                max_angle += std::abs(variables1[0] - variables2[0]);
            }
                break;
        }
    }
    return max_angle;
}
```

The idea being: assuming convex collision shapes, the convex hull approximation corresponds **exactly** to the swept volume. So, rotation is the only factor contributing to to discrepancies between the swept volume and the convex hull approximation.

Hence, I'm using a number of sub-steps directly proportional to the maximum amount of rotation in the movement, which in turn comes from the sum of rotation angles of every joint.

I still think that calling this "CCD" is problematic, since we're still using a discretization that (under-)approximates the swept volume, so I still think we **need** to document this. It's a far better under-approximation, though.

simonschmeisser, 2021-10-06:
Who *reads* documentation ...

I'm not very deep into the bullet and CCD code but for me it sounds like we could use a PR with code like yours that does some clever subdivisioning.

werner291, 2021-10-07:
```
TEST_F(BulletCollisionDetectionTester, CastHullAdversarialSwing) {

  moveit::core::RobotState state1(robot_model_);

  state1.setVariablePositions({ { "panda_joint1", 0.0 },
                                { "panda_joint2", -M_PI/2.0 },
                                { "panda_joint3", 0.0 },
                                { "panda_joint4", 0.0 },
                                { "panda_joint5", 0.0 },
                                { "panda_joint6", M_PI/2.0 },
                                { "panda_joint7", 0.0 } });

  state1.update();

  moveit::core::RobotState state2(robot_model_);

  state2.setVariablePositions({ { "panda_joint1", 0.0 },
                                { "panda_joint2", M_PI/2.0 },
                                { "panda_joint3", 0.0 },
                                { "panda_joint4", 0.0 },
                                { "panda_joint5", 0.0 },
                                { "panda_joint6", M_PI/2.0 },
                                { "panda_joint7", 0.0 } });

  state2.update();

  collision_detection::CollisionRequest req;
  collision_detection::CollisionResult res;

  cenv_->checkRobotCollision(req, res, state1, state2, *acm_);
  ASSERT_FALSE(res.collision);
  res.clear();

  // Adding the box which is not in collision with the individual states but with the casted one.
  shapes::Shape* shape = new shapes::Box(1.0, 1.0, 0.1);
  shapes::ShapeConstPtr shape_ptr(shape);

  Eigen::Isometry3d pos{ Eigen::Isometry3d::Identity() };
  pos.translation().x() = 0.0;
  pos.translation().y() = 0.0;
  pos.translation().z() = 0.8;
  cenv_->getWorld()->addToObject("box", shape_ptr, pos);

  cenv_->checkRobotCollision(req, res, state1, *acm_);
  ASSERT_FALSE(res.collision);
  res.clear();

  cenv_->checkRobotCollision(req, res, state2, *acm_);
  ASSERT_FALSE(res.collision);
  res.clear();

  moveit::core::RobotState upright(robot_model_);

  upright.setVariablePositions({ { "panda_joint1", 0.0 },
                                { "panda_joint2", 0.0 },
                                { "panda_joint3", 0.0 },
                                { "panda_joint4", 0.0 },
                                { "panda_joint5", 0.0 },
                                { "panda_joint6", M_PI/2.0 },
                                { "panda_joint7", 0.0 } });

  upright.update();

  cenv_->checkRobotCollision(req, res, upright, *acm_);
  ASSERT_TRUE(res.collision);
  res.clear();

  cenv_->checkRobotCollision(req, res, state1, state2, *acm_);
  ASSERT_TRUE(res.collision);

}
```

This seems to fail without my patch, as predicted.