moveit/moveit #2891: Servo: Apply TwistStamped in frame of header

udopa, 2021-09-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I'm using servo for path following and want to set a flexible TCP which can be offset to the known robot TCP (i.e. end of the kinematic chain, `TCP_Robot`) in Translation and Rotation. I'll call this frame `TCP_Shifted` for better explanation. In normal planning applications this is possible, for example by using a subframe of a collision object as introduced in #1439 and following issues/PRs. It would be good if I can publish a `TwistStamped` message with a `frame_id` pointing to a subframe of a collision object or any known frame in the planning scene. The latter ones could be published by a StaticTransformBroadcaster.

The motion should be executed in the given frame `TCP_Shifted`. This implies that:
- Linear motions (translation) can be executed in any frame, as long as the motion is transformed correctly (v_y could become v_x with a rotation change of 90°). This does work already.
- Angular motions (rotation) have to be executed in the given frame because otherwise the center of rotation will be wrong, and a linear motion will be introduced because of the lever between the tcp of the movegroup and the given `TCP_Shifted`. See the image for a more hands-on explanation of the problem. Currently the motion will be transformed to the controlling frame `TCP_Robot` "correctly", i.e. w_y would become w_x similar to the linear motion case. Therefore the described problem exists. 
[image]
_Image: TwistStamped with `header.frame_id=TCP_Shifted`. I want the motion to be executed in "TCP_Shifted" (desired state), but it is executed in "TCP_Robot" (current state) creating a linear motion from an angular motion and therefore moving the desired TCP even when only rotating._

This issue is related to #2866. I changed to code so that I am able to use any frame known in the scene in my `TwistStamped` commands, but I am not able to change the controlled frame to be the `TCP_Shifted`. This is possible by changing the code in servo_calcs.cpp at [L492](https://github.com/ros-planning/moveit/blob/d1e1c3d27ab33978fdcc5d7f9494afaa76496adc/moveit_ros/moveit_servo/src/servo_calcs.cpp#L492):
```cpp
planning_scene_monitor::LockedPlanningSceneRO planning_scene(planning_scene_monitor_);
const auto tf_moveit_to_incoming_cmd_frame =
    current_state_->getGlobalLinkTransform(parameters_.planning_frame).inverse() *
    planning_scene->getFrameTransform(cmd.header.frame_id);
```

**Describe the solution you'd like**
Servo in the given frame of the TwistStamped header.

**Describe alternatives you've considered**
I tried manually calculating the linear motion resulting from the rotation around the wrong center of rotation (`TCP_Shifted`) before sending the TwistStamped message to servo, but this was not accurate enough because servo then also applies a controller and my assumed velocities were inaccurate. Maybe this could also be done inside `servo_calcs.cpp` but I don't think that would be the best solution.

I also tried adding a collision object to the robot and the scene does know that there is an attached body. I used the following in [L511](https://github.com/ros-planning/moveit/blob/d1e1c3d27ab33978fdcc5d7f9494afaa76496adc/moveit_ros/moveit_servo/src/servo_calcs.cpp#L511)
```cpp
planning_scene_monitor::LockedPlanningSceneRO planning_scene(planning_scene_monitor_);
planning_scene->printKnownObjects(std::cout);
```
[image]

However, while the planning scene does recognize that there is an attached body, the joint model group used to calculate the Jacobian  [L513](https://github.com/ros-planning/moveit/blob/d1e1c3d27ab33978fdcc5d7f9494afaa76496adc/moveit_ros/moveit_servo/src/servo_calcs.cpp#L513) does not receive any updates. The current state shows that there is no object attached. [L511](https://github.com/ros-planning/moveit/blob/d1e1c3d27ab33978fdcc5d7f9494afaa76496adc/moveit_ros/moveit_servo/src/servo_calcs.cpp#L511):
```cpp
current_state_->getAttachedBodies(attached_bodies);
ROS_INFO_STREAM("attached_bodies.size()");
ROS_INFO_STREAM(attached_bodies.size());
for (auto ab : attached_bodies)
{
  ROS_INFO_STREAM(ab);
}
```
[image]

Therefore the motion is executed in `TCP_Robot`.

I think updating the joint_model_group or the robot_model would be the cleanest way, but I'm no expert in the MoveIt internal calculations and hence do not know how to do it / if it would be a bad idea. Feedback is appreciated.

If you can give me some hints on how to solve my issue and add the feature I can also have a try at implementing it myself.

Please feel free to ask if there is anything unclear.

AndyZe, 2021-10-01:
@felixvd made a good comment that also applies here:  https://github.com/ros-planning/moveit/issues/2866#issuecomment-917676690

So if you open a PR for Servo that uses getFrameTransform() instead of getGlobalLinkTransform(), I expect you could use a subframe. I believe the only changes would be in this file so it should be a pretty easy change. Looking forward to reviewing it!

AndyZe, 2021-10-01:
I see that you have already done some digging to find #2866 and [the recommendation](https://github.com/ros-planning/moveit/issues/2866#issuecomment-917676690) to use getFrameTransform. Nice.

I don't think it matters if the joint group is aware of the subframe; it just returns a Jacobian in the planning frame. Which is fine because the twist command always gets transformed to the planning frame [here](https://github.com/ros-planning/moveit/blob/049be73b8343d67ffde400fed9ed4c387c6ea9e5/moveit_ros/moveit_servo/src/servo_calcs.cpp#L470).

I'm thinking the fix may be here:

https://github.com/ros-planning/moveit/blob/049be73b8343d67ffde400fed9ed4c387c6ea9e5/moveit_ros/moveit_servo/src/servo_calcs.cpp#L493

https://github.com/ros-planning/moveit/blob/049be73b8343d67ffde400fed9ed4c387c6ea9e5/moveit_ros/moveit_servo/src/servo_calcs.cpp#L494

This code does the transform when the incoming command frame is not in the planning frame nor the EE frame. Which I think is your case. So please try `getFrameTransform()` there.

AndyZe, 2021-10-01:
^don't worry about things not being perfect on your first PR. They almost never are, for anybody  ;)

sguptasarma, 2023-05-19:
Is this still an issue (in MoveIt1)?

AndyZe, 2023-05-19:
Yes. But it's a rather rare edge case that most people won't notice

AltinayGrass, 2025-03-04:
If this topic is still open, it's time to focus on moveit2. I learned by experiencing the same problem.