moveit/moveit #2941: Is there any code examples to use approximate constraint manifold in moveit?

xibeisiber, 2021-11-03:
Hi all, I intend to do planning with approximated constraint manifold to make sure that the end effector can only rotate along the z_axis of the base link. Following the tutorial, I generated and loaded the database, but not sure know how to use it in move_group c++ interface.

Thanks for any help!

v4hn, 2021-11-03:
Why did you close your ros answers question? I just wanted to reply there only to find it's not available anymore...

> I generated and loaded the database, but not sure know how to use it in move_group c++ interface.

The move_group node loads the database on startup and registers [the name of the constraint](https://github.com/ros-planning/moveit_msgs/blob/master/msg/Constraints.msg#L4) that you defined when you generated the database.
To use it for planning you just have to set a path constraint with the same constraint name. Make sure you also add the full constraint specification there as well though because this message is still used to check transitions between the stored states.

I'll close this because I believe I fully answered the question. If you have follow-up questions please create a ros-answers question again and add my answer there.

closed 2021-11-03 by v4hn

gavanderhoorn, 2021-11-03:
> Why did you close your ros answers question? I just wanted to reply there only to find it's not available anymore...

it is now: [Is there any examples to use approximate constraint manifold in moveit?](https://answers.ros.org/question/389966).

xibeisiber, 2021-11-04:
(º﹃º ) @v4hn Thanks very much. 
I found a similar question left unanswered on answers.ros.org, so I thought you are not so active there...
The approximate constraint manifold works now. previously I didn't get what the "name" of the constraints means..
I will copy your answer to my question on answers.ros.org.

v4hn, 2021-11-04:
> I found a similar question left unanswered on answers.ros.org, so I thought you are not so active there...

can't be everywhere at once... 
it's always gambling for a good reply anyway, no matter on what platform you ask. 

> I will copy your answer to my question on answers.ros.org.

:+1: