moveit/moveit #2942: Pilz planner doesn't consider obstacles?

rhaschke, 2021-11-03:
As pointed out in https://github.com/ros-planning/moveit/issues/2741, item 1, the Pilz planners don't seem to consider obstacles.
Rather, only the post-planning trajectory validation seems to reject an invalid plan:
https://github.com/ros-planning/moveit/blob/0d7462f140e03b4c319fa8cce04a47fe3f650c60/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp#L268

As this validation can be disabled, this approach is highly questionable:
https://github.com/ros-planning/moveit/blob/cdda16fdee21f05d6a6209d59e08cc6086d9efb6/moveit_ros/planning/planning_pipeline/include/moveit/planning_pipeline/planning_pipeline.h#L100-L102

@jschleicher, can you comment on this?

v4hn, 2021-11-04:
We've talked about this a number of times in different threads.
My personal take on this by now is that we should leave this as-is unless someone finds a more severe shortcoming with the APIs due to this.

The validation is enabled by default and it should only be disabled with the behavior of the exact planner in mind. Maybe we could add a comment on that somewhere.
If we add redundant collision checking to the planner itself, we won't gain anything really, except for higher runtime.

rhaschke, 2021-11-04:
Agreed. However, we should definitely add a comment somewhere (where?) as both you (in #2741, no.1) and me stumbled over this behavior...
Actually, I suggest renaming the Pilz planner to: `pilz_industrial_motion_generator` (instead of _planner_).

jschleicher, 2021-11-04:
@rhaschke Regarding the name there have been several changes already, introducing glitches in user code every time so imho a name change breaking user launch file isn't worth the trouble. About discussions see https://github.com/ros-planning/moveit_tutorials/pull/461#issuecomment-591189071 and I remember a maintainer meeting discussion, whether it is an "incomplete planner" or "trajectory generator" or something different...

Collision checking is just executed once for runtime reasons, as @v4hn explained.

rhaschke, 2021-11-04:
Thanks for the pointer. @jschleicher, could you please add a comment about collision handling in the [Pilz tutorial](https://ros-planning.github.io/moveit_tutorials/doc/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html)?
There, it's already mentioned that the planner is actually a trajectory _generator_, but collision handling is not mentioned yet.
Thanks!