moveit/moveit #2953: MSA issues

rhaschke, 2021-11-08:
Recreating panda_moveit_config, I noticed the following issues and shortcomings:
- [ ] Writing group states doesn't ignore mimic joints as expected
- [ ] Additionally to `ros_controllers.yaml`, also read `simple_moveit_controllers.yaml` and `gazebo_controllers.yaml`
- [ ] `SRDF/relative_path` in `.setup_assistant` is not updated to `config/robot.srdf` when the file was actually written
- [ ] `simple_moveit_controllers.yaml` isn't uploaded to param server (https://github.com/rhaschke/panda_moveit_config/commit/4795be758fcd790bdfd71c4a89b99c1c1a8d4f29)

mayman99, 2021-11-08:
If no one is assigned I could work on at least one of them, maybe the second one.

rhaschke, 2021-11-08:
@mayman99, that would be great! I think the 2nd item is the most important/useful from the list. The culprit here is, that the same (ROS) controller can (but doesn't need to) show up in both
- `ros_controllers.yaml` (as type `[position|velocity|effort]_controllers/JointTrajectoryController`)
- and `simple_moveit_controllers.yaml` (as type `FollowJointTrajectory`)

These (if they share the same name) should be merged into a single item in MSA's internal controller list.

Note that `panda_moveit_config` has an empty [`ros_controllers.yaml`](https://github.com/ros-planning/panda_moveit_config/blob/cc290bccc2442495b37329c03c9cd7cbd67b1955/config/ros_controllers.yaml), because its ROS controllers are declared in an external package, namely `franka_ros`. Nevertheless, the controllers for the `SimpleControllerManager` are listed in [`simple_moveit_controllers.yaml`](https://github.com/ros-planning/panda_moveit_config/blob/cc290bccc2442495b37329c03c9cd7cbd67b1955/config/simple_moveit_controllers.yaml). The code should handle this use case as well, of course.

A major problem of MSA in general is, that the internal representation cannot cope with unexpected user changes.
For example, [`franka_control/default_controllers.yaml`](https://github.com/frankaemika/franka_ros/blob/3f06e43eff36958e3fe0640cfdcf349903429d8e/franka_control/config/default_controllers.yaml#L32) defines a controller of type `franka_control/FrankaStateController`. MSA doesn't know about this type (and doesn't need to know). However, it should exactly reproduce this controller entry. Same for YAML [refs](https://github.com/frankaemika/franka_ros/blob/3f06e43eff36958e3fe0640cfdcf349903429d8e/franka_control/config/default_controllers.yaml#L29) and [anchors](https://github.com/frankaemika/franka_ros/blob/3f06e43eff36958e3fe0640cfdcf349903429d8e/franka_control/config/default_controllers.yaml#L1) used in there. 
To handle these correctly, the YAML tree shouldn't be converted into an internal representation at all. Rather, MSA should operate on the YAML tree directly, via a corresponding [`QAbstractItemModel`](https://doc.qt.io/qt-5/qabstractitemmodel.html) wrapping the YAML representation. A quick search on Google didn't give hits for such a data model, though :disappointed:

mayman99, 2021-11-09:
@rhaschke thanks for the clarification, but just to make sure I've the right idea
> Note that panda_moveit_config has an empty ros_controllers.yaml, because its ROS controllers are declared in an external package, namely franka_ros. Nevertheless, the controllers for the [...]

So we would allow the user to load a `controllers` yaml through the controllers screen, so the controllers yaml wouldn't have to always be in the same package for the MSA to work, right?

> To handle these correctly, the YAML tree shouldn't be converted into an internal representation at all. Rather, MSA should operate on the YAML tree directly, via a corresponding QAbstractItemModel [...]

If we get to the MSA to operate on a wrapped YAML tree, it would be a nice upgrade, it really needs this flexibility so it breaks fewer config packages.

rhaschke, 2021-11-09:
> So we would allow the user to load a `controllers` yaml through the controllers screen, so the controllers yaml wouldn't have to always be in the same package for the MSA to work, right?

I didn't think of offering an option to load external controller yamls. I wanted to point out that two different use cases exist:
- MoveIt defining ros_controllers.yaml and simple_moveit_controllers.yaml - with (partially) identical controller names
- MoveIt only defining simple_moveit_controllers.yaml, _referring_ to some external controllers.yaml

> If we get to the MSA to operate on a wrapped YAML tree, it would be a nice upgrade, it really needs this flexibility so it breaks fewer config packages.

I consider this nice to have. However, it's important to consider existing user settings for `default` and `action_ns`.
This can also be achieved w/o writing a YAML-wrapping item model.

mayman99, 2021-12-22:
Hey, so an update about my progress:
The MSA can read any arbitrary format of `yaml` files, parse them and store it in an abstract data structure. lets call it `yaml_struct`.
The MSA can still write the same exact file it read (refs and anchors also work with this implementation).

This was the current status, and here is how I plan I should proceed:
Using [`QAbstractItemModel`](https://doc.qt.io/qt-5/qabstractitemmodel.html) is possible, but after researching this would take a lot of resources to be implemented.

So for `ros_controllers` we could replace the functions for `yaml_reading` that we currently use with this new implementation for reading all yaml files, and modify the `controllers widget` to operate on the new structure instead of the old `ControllersConfig`.
What do you think?

rhaschke, 2021-12-22:
Do you use https://github.com/jbeder/yaml-cpp and its data structure(s)?
In general, I agree that operating on the yaml data structures should do the job. How do you plan to handle the multitude of controller config files? For each of them, you will have a yaml data structure, which then need to be combined into the single MSA widget. There you need to keep track for all entries where they are originating from.