moveit/moveit #2956: Make max_effort settable in the MoveIt Simple Controller Manager

rickstaa, 2021-11-11:
**Is your feature request related to a problem? Please describe.**

I think the [moveit_simple_controller_manager](https://github.com/ros-planning/moveit/tree/master/moveit_plugins/moveit_simple_controller_manager) simple can be improved such that users can easily specify the MaxEffort that is set on the
`GripperCommand` message. This will allow users to grasp an object and move the gripper to the desired position (see https://github.com/frankaemika/franka_ros/pull/173#issuecomment-965098234).

#### TODOS
- Specify the GripperCommand max effort as a ROS parameter in the GripperCommand declaration.
- Allow switching controllers for different (types of) trajectories (see https://github.com/ros-planning/moveit/pull/1832).
- Add effort field to [motion_planning_rviz_plugin](https://github.com/ros-planning/moveit/blob/master/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_display.cpp) when group uses GripperCommand action.

Allow specification of controller names for trajectories ros-planning/moveit#1832

v4hn, 2021-11-11:
For completeness, the max_effort field currently depends on [the effort set for the last waypoint of the trajectory to be executed](https://github.com/ros-planning/moveit/blob/master/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager/gripper_controller_handle.h#L132-L133).
That behavior should definitely be retained and is more flexible than what you propose.
But it's very true that there is no interface to specify the effort for this waypoint when you plan through the `MoveGroup` action.
It could make sense to add a `max_effort` field to the generic action requests (or PlanningOptions) and use this field for all waypoints if specified.

rickstaa, 2021-11-11:
> For completeness, the max_effort field currently depends on the effort set for the last waypoint of the trajectory to be executed.
That behavior should definitely be retained and is more flexible than what you propose.

@v4hn I was planning to read the default `max_effort` from the parameter server and only apply it when no `effort` is set anywhere in the send Trajectory. Meaning if somebody sets an effort for one of the waypoints the parameter is no longer used.

> But it's very true that there is no interface to specify the effort for this waypoint when you plan through the MoveGroup action.
It could make sense to add a max_effort field to the generic action requests (or PlanningOptions) and use this field for all waypoints if specified.

This was how I was planning to implement it.

> EDIT(v4hn): a group does not use an action, a set of joints do. So this feature would break abstractions.

Is there a way that we can add the max effort to the `motion_planning_rviz_plugin` without breaking the abstractions? Or would you suggest dropping this TODO and only providing it through the `MoveGroup` action?

v4hn, 2021-11-11:
> I was planning to read the default `max_effort` from the parameter server and only apply it when no `effort` is set anywhere in the send Trajectory.

Sounds very useful. :+1:

> > It could make sense to add a max_effort field to the generic action requests (or PlanningOptions) and use this field for all waypoints if specified.
> 
> Is there a way that we can add the max effort to the `motion_planning_rviz_plugin` without breaking the abstractions? Or would you suggest dropping this TODO and only providing it through the `MoveGroup` action?

It could be useful in the GUI. If we would add such a generic max_effort field to the action, we could also fill it in from the GUI.
But then again, shouldn't the generic interface actually allow to specify *different* maximum efforts for different joints?
In that case a bit more work would be needed also with the GUI.

rickstaa, 2021-11-11:
> It could be useful in the GUI. If we would add such a generic max_effort field to the action, we could also fill it in from the GUI.
But then again, shouldn't the generic interface actually allow to specify *different* maximum efforts for different joints?
In that case a bit more work would be needed also with the GUI.

Good point. In that case, it would be similar to the [joints](https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html?highlight=null%20space#moving-joints-or-in-null-space) tab. I will try to approximate how much work adding this is. so we can check if adding it justifies the development time spend on it.

rickstaa, 2021-12-07:
@rhaschke and @v4hn, I will not implement the RVIZ effort GUI since this feature requires more work after inspection than I can currently dedicate to this pull request. This might be something to add in the future.

rickstaa, 2021-12-07:
@rhaschke I think the main functionality of https://github.com/frankaemika/franka_ros/pull/173#issuecomment-965098234 was implemented in #2984. Please let me know if it is what you had in mind.

You also mentioned something about controller switching in https://github.com/frankaemika/franka_ros/pull/173#issuecomment-965098234. I, however, was not sure what you had in mind. It looks like that is a distinct feature not related to the gripper control we requested https://github.com/frankaemika/franka_ros/pull/173#issuecomment-965012307.

rickstaa, 2021-12-15:
@rhaschke Now that https://github.com/ros-planning/moveit/pull/2984#event-5769897890 has been merged, I think we can close this issue. The only TODO that is open is the one about the controller switching (see https://github.com/ros-planning/moveit/issues/2956#issuecomment-988231607).

rhaschke, 2022-01-14:
Closed via #2984. 
The controller switching I was referring to was already implemented in https://github.com/ros-planning/moveit/pull/1832.
The core idea is to explicitly specify which controller (names) to use for specific sub trajectories of a longer plan. This way, one can switch between e.g. GripperCommand and JointTrajectoryControl.

closed 2022-01-14 by rhaschke