moveit/moveit #2966: Robot not displayed in Moveit Setup Assistant Noetic

j23saw, 2021-11-18:
### Description
Hello,
I tried to use moveit setup assistant with open manipulator for learning but after loading it didn't show up in the display so I tried it with the franka_description and have the same issue. I saw similar issue for ros-indigo and did try exporting export $LC_NUMERIC but it didn't change anything.

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04 (WSL2)
* Source or Binary build? Binary
* If binary, which release version? 1.1.5
* If source, which branch?

### Steps to reproduce
I cloned the franka_description while following the setup-assistant tutorial on https://ros-planning.github.io/ and loaded the panda_arm_.urdf.xacro file. The setup happens and all the collisions are shown in the second tab but it is not displayed in the visual panel.

### Backtrace or Console output
[backtrace](https://gist.github.com/j23saw/1cf19ef05b9ba5e98ef647e57fb750cd)

rhaschke, 2021-11-18:
Can you see a robot / anything in rviz, in general? Maybe, the problem is related to your WSL2 setup.

j23saw, 2021-11-18:
Oh no I can't see it in rviz too, you may be right it might be related to my WSL2 setup. Any pointers to debug this problem?

simonschmeisser, 2021-11-18:
First please check in rviz if you can see any of the generic stuff and also if the `RobotDisplay` works

j23saw, 2021-11-19:
I can't see any RobotDisplay option here in rviz

rhaschke, 2021-11-21:
`RobotModel` display?

j23saw, 2021-11-21:
There's no RobotModel display but I can see "Scene Robot"

rhaschke, 2021-11-21:
[image]

j23saw, 2021-11-21:
Ah yes, I added RobotModel but still it can't display the actual 3d model, only collision is visible

rhaschke, 2021-11-21:
Maybe you are running into this issue: https://answers.ros.org/question/272692/rviz-dows-not-display-model-urdf-tutorial?
I thought this was fixed...

rhaschke, 2021-11-21:
Collision and visual models of franka_description use different approaches: visual models are .dae files, collision models are basic geometry primitives, directly rendered by rviz. Maybe you have an issue with the `assimp` package as well? This is used to load meshes.

simonschmeisser, 2021-11-21:
> Maybe you are running into this issue: https://answers.ros.org/question/272692/rviz-dows-not-display-model-urdf-tutorial? I thought this was fixed...

Yes, I fixed that some years ago and they actually also tried setting LC_NUMERIC so that should not be the issue here

simonschmeisser, 2021-11-21:
@lilustga do you have any experience with WSL2?

Are there no error messages at call on the command line?

lilustga, 2021-11-21:
@simonschmeisser I do have experience with WSL2 and ROS. Have not seen this problem before though

mashuai191, 2021-12-21:
looks quite similar with my reported  #3006 , and my env is also WSL2, on win11