moveit/moveit #2970: null-space sliders can leave null-space

v4hn, 2021-11-18:
Playing around with the GUI today, I noticed [this breakage in the joint sliders' null-space exploration](https://tams.informatik.uni-hamburg.de/people/goerner/files/pr2-explore-null-space-slider.mp4).

For some reason dragging the null-space slider can leave the null-space and start moving the end effector away from the picked pose.
I'm not sure whether this is an artifact due to joint limits, accumulated noise (which seems to be bounded?) or something else entirely...

rhaschke, 2021-11-19:
This is an expected phenomenon if you take too large steps in the linear approximation of the nonlinear nullspace.

v4hn, 2021-11-19:
True, I wasn't aware this is your interpretation of the slider displacement. The clear shortcoming of this is that you can't move fast in a more accurate approximation, but only veeeeery slowly.

Why should the slider motion directly translate into step-width of single update steps? Would you be opposed to interpret the motion as, e.g., distance in joint space with some higher sampling rate for the linear approximation before? Ideally we could define a trust region to vary the step-width dynamically, though I'm not sure that is mandatory.

rhaschke, 2021-11-19:
Sure, everything is possible...