moveit/moveit #2974: can i use curses library in moveit_commander_cmdline.py source?

EmadDeve20, 2021-11-24:
# Why curses?

this is good library and i searched about this and now i know to this library is standard and we can use it in  ```Python 2.7.18```!

## curses for what?

so in ```moveit_commander_cmdline.py``` source, i see a class with ```bcolors``` name for change font color!
and in ```print_message``` function is used it!
we can use easily **curses** library for this!

## I am Right?

so i am not sure about this issue because i think this class can better than **curses** library!
Did you have a specific reason for not using **curses** library?

rhaschke, 2021-11-24:
Do you suggest using the `curses` library instead of some proprietary methods/classes? If so, we welcome your PR!

EmadDeve20, 2021-11-25:
# Report 
Hi @rhaschke sorry for i late!
the ```curses``` library is so big and hard for using but for change font color in python i use another standard library with ```pygments``` name.

and my branch is ```use_standard_library_in_moveit_commander_cmdline.py```

and my code is:
```
7- from pygments import console
...
77- def print_message(level, msg):
78-     if level == MoveGroupInfoLevel.FAIL:
79-         print(console.colorize("red", msg))
80-    elif level == MoveGroupInfoLevel.WARN:
81-         print(console.colorize("yellow", msg))
82-     elif level == MoveGroupInfoLevel.SUCCESS:
83-         print(console.colorize("green", msg))
84-     elif level == MoveGroupInfoLevel.DEBUG:
85-         print(console.colorize("cyan", msg))
86-     else:
87-         print(msg)
```

## problem in this case
so the big problem for ```pygments``` library is, it`s not in the python2!
### solution
we has two solution for this problem!
one: we can use if and some try/catch for handling this problem!
two: This is not a hot and public relations change and we can keep the code as before :D

EmadDeve20, 2021-11-25:
so i think we no need this issue!

closed 2021-11-25 by EmadDeve20