moveit/moveit #2978: can i move joint my visual robot in rviz with moveit API? (Only joints without using the design and execution, that is, before them and real time)

EmadDeve20, 2021-11-28:
I am a developer to create a user interface to control the robot!
I use the visual robot directly from rviz in my Windows UI.
My robot has 6 joints and in my interface I have 6 sliders to control the joints!

I talked about this research but I did not see any answer!

[image]

I apologize for asking my questions here! I asked in some forums but got no answer and said I would try my luck here.

this is a part of my UI:
[image]

Is there a way I can shake the joints using the slider widget?
Can **moveit** or has another solution? I did a lot of research on this and saw the complete **moveit** document and I don't think **moveit** does that right?

rhaschke, 2021-11-28:
There are multiple options to do so and I'm surprised that you couldn't find them in the docs!
- With rviz only, you can use multiple RobotModel displays loading their robot_description from different ROS namespaces.
  These should use different link and joint prefixes (e.g. via xacro) to distinguish different instances of the same type.
- MoveIt offers the `RobotState` display, which also allows attaching objects to individual links and doesn't rely on TF to display the links.

closed 2021-11-28 by rhaschke

EmadDeve20, 2021-12-08:
Hi @rhaschke I shared this link with my team and my team guessed that maybe you did not understand what I meant!
my question isi want to use slider for move the robot joints. (without plan and execute or i think without moveit)
this is posible?

rhaschke, 2021-12-08:
I think you are simply looking for http://wiki.ros.org/joint_state_publisher_gui.