moveit/moveit #2979: boost: mutex lock failed when robot_model_loader goes out of scope

captain-yoshi, 2021-12-02:
### Description
When the `robot_model_loader` object goes out of scope and there is a path constraint, an exception is caught in the planning request adaptors and the plan fails.

Quick example
```cpp
moveit::core::RobotModelPtr model;
{
    robot_model_loader::RobotModelLoader loader(ROBOT_DESCRIPTION);
    model = loader.getModel();
}
...
// req contains a path constraint
pipeline->generatePlan(scene, req, res);
```
I though I would not need the loader once I got the model. If the loader is not destroyed everything works as expected. This only happens with the OMPL pipeline. CHOMP and STOMP does not suffer from this (They reject the plan).

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* If source, which branch? master fd36674cc327962aaf27925ddf1ba9c6a8667d35

### Steps to reproduce
Clone this miniaml example [repo](https://github.com/captain-yoshi/moveit_debug/tree/master), checkout branch `motion-plan-request-path-constraints` and run:
```bash
roslaunch moveit_debug example.launch
```
### Expected behaviour
Not to fail when adding a path constraint.

### Actual behaviour
Planning fails (throws from the ompl interface). See GDB backtrace below.

### Backtrace or Console output
Console
```terminal
[ERROR] [1638416577.208121136]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.208477660]: Exception caught executing *next* adapter 'Fix Start State In Collision': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.213055804]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.213494690]: Exception caught executing *next* adapter 'Fix Start State Bounds': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.219902171]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.220173946]: Exception caught executing *next* adapter 'Fix Start State In Collision': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.224300961]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.224525320]: Exception caught executing *next* adapter 'Fix Workspace Bounds': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.230725116]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.230950218]: Exception caught executing *next* adapter 'Fix Start State In Collision': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.235232976]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.235446598]: Exception caught executing *next* adapter 'Fix Start State Bounds': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.242837062]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638416577.243122355]: Exception caught executing *next* adapter 'Fix Start State In Collision': boost: mutex lock failed in pthread_mutex_lock: Invalid argument

```
GDB [backtrace](https://gist.github.com/captain-yoshi/cd2e40dc2dcd4af80b512e8af9b38320) when removing try catch from planning pipeline and the planning request adapter.

The boost mutex lock seems to fail at this [ompl section](https://github.com/ros-planning/moveit/blob/master/moveit_planners/ompl/ompl_interface/src/parameterization/work_space/pose_model_state_space.cpp#L56).

rhaschke, 2021-12-02:
You need to keep the RobotModelLoader around. Destroying it before the model, should give you some pluginlib warnings in the console as well, doesn't it?

captain-yoshi, 2021-12-02:
No pluginlib warnings as shown below. I see, the `KinematicsPlugin` is needed and resides in the loader. This is strange because I was benchmarking planners without path constraints for a couple of months without issues. Feel free to close this issue.
```terminal
NODES
  /
    example (moveit_debug/run_example)

ROS_MASTER_URI=http://localhost:11311

process[example-1]: started with pid [695053]
[ INFO] [1638455360.930639518]: Loading robot model 'panda'...
[ WARN] [1638455361.000232265]: SEVERE WARNING!!!
Attempting to unload /home/captain-yoshi/ws/ros/gsoc_ws/devel/lib//libmoveit_kdl_kinematics_plugin.so
while objects created by this library still exist in the heap!
You should delete your objects before destroying the ClassLoader. The library will NOT be unloaded.
[ INFO] [1638455361.058755508]: Using planning interface 'OMPL'
[ INFO] [1638455361.062382558]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1638455361.062806341]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1638455361.063156654]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1638455361.063556955]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1638455361.063917901]: Param 'jiggle_fraction' was not set. Using default value: 0.02
[ INFO] [1638455361.064277529]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1638455361.064383595]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1638455361.064423823]: Using planning request adapter 'Resolve constraint frames to robot links'
[ INFO] [1638455361.064462532]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1638455361.064474018]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1638455361.064503843]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1638455361.064515600]: Using planning request adapter 'Fix Start State Path Constraints'
[ERROR] [1638455361.141492343]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638455361.141791833]: Exception caught executing *next* adapter 'Fix Start State In Collision': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638455361.144816977]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638455361.145073111]: Exception caught executing *next* adapter 'Fix Start State Bounds': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638455361.150067130]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638455361.150287559]: Exception caught executing *next* adapter 'Fix Start State In Collision': boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ERROR] [1638455361.153807135]: Exception caught executing *final* adapter 'Fix Start State Path Constraints': boost: mutex lock failed in pthread_mutex_lock: Invalid argument

```

closed 2021-12-02 by rhaschke