moveit/moveit #2985: Scene format breakage between 1.1.5 and 1.1.6 in released distro (Noetic)

wxmerkt, 2021-12-08:
Hi, we are downstream users of the `.scene` format for describing collision scenes. Today, I've noticed that with the latest MoveIt from binary on Noetic, our unit tests and operational code that relied on scene loading started to fail (indicating a parsing error for the translation component).

It turns out that this is due to #2037 and documented there in the migration guide for Noetic:
> - Planning scene geometry text files (`.scene`) have changed format. Add a line `0 0 0 0 0 0 1` under each line with an asterisk to upgrade old files if required.

The issue is that #2037 was released into binaries last month as part of https://github.com/ros/rosdistro/pull/31147 (v1.1.6), following an earlier released version (v1.1.5) that used the old API.

One solution would be to update all `.scene` files following the migration guide in our code, with other end-users having to do the same (with the implication that we'd have to maintain separate `.scene` files for Melodic). However, since this targeted an already released distro, I wonder if adding backwards-compatibility (check number of lines for this object, if too few, use default object pose) was on option? Alas, I don't have cycles to contribute this myself.

Many thanks,
Wolfgang

rhaschke, 2021-12-08:
Dear Wolfgang, in order to integrate new features into Noetic we have to go for breaking changes sometimes. We couldn't make any progress in ROS1 otherwise, as there will be no follow-up distro after Noetic. We are sorry for the trouble this introduces. 
Regarding backwards compatibility: If you can come up with a nice solution, we will be happy to merge it. I cannot promise anybody from the maintainer team will have time to look into this anytime soon. Most maintainers moved on to ROS2 development and we have lots of pending PRs open here already...

simonschmeisser, 2021-12-08:
I would really appreciate a PR that adds a new scene file format based on something sane. yaml would be an obvious candidate as that is likely already a dependency

>  Most maintainers moved on to ROS2 development and we have lots of pending PRs open here already...
PRs added to ROS1 will be forward ported to ROS2 by Picknick so it's not like you would invest in a dead horse ...

rhaschke, 2021-12-08:
I didn't want to raise the impression that investing into ROS1 MoveIt is useless :wink: I just wanted to clarify that the manpower available for ROS1 maintenance has drastically reduced - unfortunately.

closed 2021-12-10 by v4hn

wxmerkt, 2021-12-10:
Thank you for merging the fix @v4hn. What are the current plans for backporting & a Noetic release to fix it?

v4hn, 2021-12-10:
noetic is synced with master at the moment, so it landed already.

I'm not aware of a scheduled release, but we can probably cut one soon.

wxmerkt, 2021-12-10:
That would be great, thank you - if we can make the next Noetic sync that'd be ideal.

wxmerkt, 2021-12-15:
Friendly ping - any thoughts / plans for the release to Noetic?

PR #2986 (2021-12-10): Hotfix: Add backwards compatibility for old scene serialization format
This PR adds backwards compatibility for the old scene serialization format to `PlanningScene` following #2037, cf. the regression report for Noetic in #2985. It first checks which version of the syntax is provided (by checking the line following the start of a new object - previously the number of shapes, now the translational component of the pose) and deactivates the subframe-poses if using the old format, as well as maintains compatibility by resetting the pose to identity.

@felixvd @rhaschke Can you please review this [... cut]