moveit/moveit #2989: sync-noetic-devel can break when two PRs are merged close to each other

v4hn, 2021-12-10:
I just received [a failed action run](https://github.com/ros-planning/moveit/actions/runs/1563785198) for the sync action and was wondering for a while what went wrong.
Turns out I merged 9d78be4705fad661a9cd2ef90b0f9d933d60b325 and 9bfb2c73cf70acef7ba9d127698831d3fe14bd04 very close to each other and the wrong one pushed first:

```
To https://github.com/ros-planning/moveit
 ! [rejected]        master -> noetic-devel (non-fast-forward)
error: failed to push some refs to 'https://github.com/ros-planning/moveit'
hint: Updates were rejected because a pushed branch tip is behind its remote
hint: counterpart. Check out this branch and integrate the remote changes
hint: (e.g. 'git pull ...') before pushing again.
hint: See the 'Note about fast-forwards' in 'git push --help' for details.
```

I don't want to experiment with this right now, but I guess
https://github.com/ros-planning/moveit/blob/master/.github/workflows/ci.yaml#L169
can be replaced by something like
```
git switch noetic-devel
git merge --ff-only master
git push origin noetic-devel
```
?
@rhaschke

rhaschke, 2021-12-10:
The key problem was probably that the wrong `noetic-devel` branch was fetched from remote. I'm not convinced that your proposal will help for this. Rather, we would need to fetch again...
Did you manually forwarded noetic-devel now? Maybe just re-triggering the CI job would have helped?
I'm not sure it's worth to put effort into fixing this...

v4hn, 2021-12-11:
no, I did not push by hand. 
I expect the job of the second mentioned commit (which suceeded) did the trick.

closed 2021-12-11 by rhaschke