moveit/moveit #2992: RoboStack CI build fails

rhaschke, 2021-12-13:
... with [following error](https://github.com/ros-planning/moveit/runs/4503122842?check_suite_focus=true):
`ninja: error: '/home/runner/micromamba/envs/test/lib/libtinyxml2.a', needed by 'devel/lib/python3.8/site-packages/pymoveit_core.cpython-38-x86_64-linux-gnu.so', missing and no known rule to make it`

`ninja: error: '/Users/runner/micromamba/envs/test/lib/libtinyxml2.a', needed by 'devel/lib/python3.8/site-packages/pymoveit_core.cpython-38-darwin.so', missing and no known rule to make it
`

Looks like tinyxml2 was removed from micromamba?

@wolfv, didn't you fix the micromamba version:
https://github.com/ros-planning/moveit/blob/304ae6acdf9bb3994ad7f7315c23048ca6c72f7e/.github/workflows/robostack.yaml#L27

What else should be fixed to a specific version?

wolfv, 2021-12-13:
tinyxml2 wasn't removed from micromamba, but we made the `tinyxml2` package emit shared libraries (instead of static ones): https://github.com/conda-forge/tinyxml2-feedstock/pull/12

Unfortunately it seems that some CMake files in the ROS ecosystem probably hard-code the `tinyxml2.a` file (instead of calling `find_package(tinyxml2)` again. We can fix this (temporarily) by locking down the `tinyxml2` version to the previus one.

cc @Tobias-Fischer I think now we definitely need to rebuild noetic :)

wolfv, 2021-12-13:
I pushed a PR #2993

rhaschke, 2021-12-13:
Fixed via #2993.

closed 2021-12-13 by rhaschke