moveit/moveit #2994: Duration is out of dual 32-bit range

fzoric8, 2021-12-13:
### Description
I'm sometimes having issue during path planning. After executing moveGroup->execute(), during execution I get following error: 
```
[ERROR] [1639416096.145879252]: Dynamic exception type: std::runtime_error
std::exception::what: Duration is out of dual 32-bit range

```

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source 
* Branch: `noetic-devel` 
* Schunk lwa4p

### Steps to reproduce
I'm currently not able to provide scripts for reproducing error, however this error happens sometimes, not always. It's quite hard to determine when and why it happens. I've tried rebuilding whole ROS workspace and I'm sure I'm using ROS `noetic` and  have adequate environment. Interesting thing is that it happens only occasionaly, and it's probably related to time duration of certain trajectory (shorter planning time, easier fail). 

Any help would be much appreciated. Even how te get verbose output to be able to better locate source of the error. 
Thank you

rhaschke, 2021-12-13:
This usually occurs when using an uninitialized duration or calculating a negative duration. Please try to narrow the issue down.

fzoric8, 2021-12-14:
Closing this up because it's not related to `moveit`. 

I was looking how to get better outputs and found [this](http://wiki.ros.org/rosconsole#Console_Output_Formatting). 
I set up rosconsole output formatting environment variable as follows: 

```
export ROSCONSOLE_FORMAT='[${severity}] [${time}] [${node}]: ${message}'
```
Maybe helps someone. 
Thank you!

closed 2021-12-14 by fzoric8