moveit/moveit #2998: How to receive position of closest point to robot (even without being in contact)

kmerckae, 2021-12-14:
Hello everyone, 

I have already used the [checkCollision](http://docs.ros.org/en/noetic/api/moveit_core/html/cpp/classplanning__scene_1_1PlanningScene.html#a4ebcc755346e443319104e8b964f8faa) function to see if there is a collision between a given robot state and the environment and in case there is no collision, to see what the _distance_ is to the closest point in the environment. 

However, I would also like to receive the _position_ of the point in the environment that is the closest to a given robot state even when none of the robot links are in contact with the environment. 

Can someone explain to me how I can get this data? 

Thanks a lot in advance! 
Kelly

simonschmeisser, 2021-12-14:
Hi there,

names might be slightly confusing here but I think you can get your data with `checkCollision`. Actually `distanceToCollision` is just a convenience wrapper around the more general `checkCollision` where you can define the exact information you need in the `CollisionRequest` and read the results from the `CollisionResults`.

Does that help? Please also have a look at the source code of `PlanningScene`, those wrapper parts are quite readable

kmerckae, 2021-12-15:
Thank you for your quick reply! 

I had already taken a look into `CollisionRequest`  and `CollisionResults`, but I think the names are indeed confusing. 

I understand that when I set `collision_request.contacts=true` , `collision_result.contact` will give me the ContactMap (map returning the pairs of body ids in contact, plus their contact details) and as such I will get the positions in the environment and on the robot links where the contacts occur. 

My question is if I can also receive a non-empty ContactMap _when there are no collisions_, such that while `collision_result.collision=0` the ContactMap would give me the position of the closest points instead of the position of the points in collision.

kmerckae, 2022-01-26:
In the meantime I found out how to find the positions of the closest points in the environment to all the robot links _even_ when the robot is not colliding with any obstacles. 

As @simonschmeisser said, this is indeed possible with the `checkCollision` function. However, if you want to receive the positions of the closest points in case the robot is not in collision with any obstacle, then the active collision detector has to be changed from FCL (default in MoveIt) to Bullet, see [Using Bullet for Collision Checking](https://ros-planning.github.io/moveit_tutorials/doc/bullet_collision_checker/bullet_collision_checker.html). Thereby, you have to set `CollisionRequest.contacts=true` and `CollisionRequest.distance=true`.  This is also mentioned in https://github.com/ros-planning/moveit/issues/1467#issuecomment-641632111. 

In the figure below you can see that there are two boxes (table and wall) and two cylinders added to the planning scene. I modified the [Allowed Collision Matrix](https://ros-planning.github.io/moveit_tutorials/doc/planning_scene/planning_scene_tutorial.html#modifying-the-allowed-collision-matrix) so that robot self-collisions and collisions between the planning scene and panda_link0 and panda_link1 are not taken into account. 
By using the checkCollision function with the CollisionRequest.distance and CollisionRequest.contacts set to true and the CollisionRequest.max_contacts_per_pair set to 1, I received the ContactMap via CollisionResult.contacts, see also [CollisionResult](http://docs.ros.org/en/noetic/api/moveit_core/html/cpp/structcollision__detection_1_1CollisionResult.html). The second elements of this ContactMap is a [Contact struct](http://docs.ros.org/en/noetic/api/moveit_core/html/cpp/structcollision__detection_1_1Contact.html), which gives among other things the `nearest_points` vector. 
Saving these nearest_points vectors for all robot links wrt all obstacles in the planning scene resulted in the green points and red arrows in the figure below. 

[image]

closed 2022-01-26 by kmerckae