moveit/moveit #3004: Servo server  UR5 not moving

stevedanomodolor, 2021-12-18:
Ubuntu 20, moveit ros noetic
I have being trying to get the pose tracking in the moveit servo example  to work but the robot does not seems not to move. The servo server seems to be sending joint position values but for some reason it is not moving(it is not changing value at all, it keeps sending the same joint values). Can someone help me please. 

Also, my joint state node includes other link of the whole robot, is that a problem?
[image]

controller screen
[image]

I am using the pose_tracking example and my launch file is 

I added a marker to see the goal in the pose_tracking example code 
```

  target_pose.pose.orientation = current_ee_tf.transform.rotation;
  ROS_INFO("Pose: %f %f %f",target_pose.pose.position.x,target_pose.pose.position.y,target_pose.pose.position.z );

  // Modify it a little bit
  target_pose.pose.position.x -= 0.1;
  target_pose.pose.position.z -= 0.1;

  // resetTargetPose() can be used to clear the target pose and wait for a new one, e.g. when moving between multiple
  // waypoints
  tracker.resetTargetPose();

  // Publish target pose
  target_pose.header.stamp = ros::Time::now();
  target_pose_pub.publish(target_pose);

  // Run the pose tracking in a new thread
  std::thread move_to_pose_thread(
      [&tracker, &lin_tol, &rot_tol] { tracker.moveToPose(lin_tol, rot_tol, 0.1 /* target pose timeout */); });
      visualization_msgs::Marker marker;
      marker.header.frame_id = "ur5_arm_base_link";
      marker.header.stamp = ros::Time();
      marker.ns = "my_namespace";
      marker.id = 0;
      marker.type = visualization_msgs::Marker::SPHERE;
      marker.action = visualization_msgs::Marker::ADD;
      marker.pose.position= target_pose.pose.position;
      marker.pose.orientation.x = 0.0;
      marker.pose.orientation.y = 0.0;
      marker.pose.orientation.z = 0.0;
      marker.pose.orientation.w = 1.0;
      marker.scale.x = 0.1;
      marker.scale.y = 0.1;
      marker.scale.z = 0.1;
      marker.color.a = 1.0; // Don't forget to set the alpha!
      marker.color.r = 0.0;
      marker.color.g = 1.0;
      marker.color.b = 0.0;
      //only if using a MESH_RESOURCE marker type:
  ros::Rate loop_rate(50);
  for (size_t i = 0; i < 6; ++i)
  while(ros::ok())
  {
    // Modify the pose target a little bit each cycle
    // This is a dynamic pose target
    // target_pose.pose.position.z -= 0.04;
    target_pose.header.stamp = ros::Time::now();
    tracker.getCommandFrameTransform(current_ee_tf);
    ROS_INFO("Pose: %f %f",target_pose.pose.position.z, current_ee_tf.transform.translation.z );

    target_pose_pub.publish(target_pose);
    vis_pub.publish( marker );

    loop_rate.sleep();
  }
```

[... 268 lines cut ...]
       something strange when unpausing it
  -->
  <node name="spawn_gazebo_model" pkg="gazebo_ros" type="spawn_model"
        args="-urdf -param robot_description -model ur5_husky_robotiq_85f
              -J ur5_arm_shoulder_lift_joint -1.1
              -J ur5_arm_elbow_joint -1.57
              -J ur5_arm_wrist_2_joint 1.57"
        output="screen" />

  <!-- <include file="$(find husky_ur5_robotiq_85)/launch/husky_ur5_robotiq_85f_controller.launch" /> -->

  <rosparam file="$(find husky_ur5_robotiq_85)/config/husky_controller.yaml" command="load" />
  <rosparam file="$(find husky_ur5_robotiq_85)/config/ur5_robotiq_85_controller.yaml" command="load" />
  <node name="ros_control_controller_manager" pkg="controller_manager" type="spawner" respawn="false" output="screen" args=" joint_state_controller ur5_joint_group_position_controller gripper_controller" />
  <node name="ros_control_controller_manager1" pkg="controller_manager" type="controller_manager" respawn="false" output="screen" args="load ur5_trajectory_controller  " />
  <node name="base_controller_spawner" pkg="controller_manager" type="spawner" args="husky_joint_publisher husky_velocity_controller"/>
  <!-- TF -->
  <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">
    <param name="publish_frequency"  type="double" value="50" />
  </node>

  <!-- <node type="rviz" name="rviz" pkg="rviz" args="-d $(find husky_ur5_robotiq_85)/rviz/husky_ur5_robotiq_85.rviz" /> -->
</launch>

```

AndyZe, 2021-12-19:
It could be that your   "joint constraint" values are too high. 0.1 rads is high:

```
constraints:
    ...
    ur5_arm_shoulder_pan_joint: {trajectory: 0.1, goal: 0.1}
```

AndyZe, 2021-12-19:
Please check that the `position_controllers/JointTrajectoryController` is stopped but the `position_controllers/JointGroupPositionController` is running. They're mutually exclusive. IIRC the command for that is `rosrun controller_manager list_controllers`

AndyZe, 2021-12-19:
Do you see any warnings in terminal about the robot starting in singularity, or anything else?

AndyZe, 2021-12-19:
Also check the output of `/servo/status`:  `rostopic echo /servo/status`

stevedanomodolor, 2021-12-20:
I have checked, it seems to be working now, I started working when i switch to effortcontroller/joint......., for some reason position_controller was not working :(

AndyZe, 2021-12-20:
Good to head. Maybe that means the Gazebo PID gains weren't set correctly.

closed 2021-12-20 by AndyZe