moveit/moveit #3006: no robot seen when i run the moveit tutorial

mashuai191, 2021-12-20:
### Description
i start to learn by going through this link https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html, but cannot see the robot mentioned here - "You should now see the Panda robot in RViz:"

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04 on WSL2
* Source or Binary build?
* If binary, which release version?
* If source, which branch? source as the link suggested

### Steps to reproduce
first go through this: https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html
then go through this: https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

You can start planning now!

[ INFO] [1640013940.964509939]: rviz version 1.14.13
[ INFO] [1640013940.964564902]: compiled against Qt version 5.12.8
[ INFO] [1640013940.964572580]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1640013940.971327842]: Forcing OpenGl version 0.
[ INFO] [1640013943.931013498]: Stereo is NOT SUPPORTED
[ INFO] [1640013943.931108535]: OpenGL device: D3D12 (NVIDIA GeForce MX450)
[ INFO] [1640013943.931163572]: OpenGl version: 3.1 (GLSL 1.4).
[ INFO] [1640013965.344652355]: Loading robot model 'panda'...
[ WARN] [1640013965.344969704]: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1640013965.345053534]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ INFO] [1640013965.423813077]: Starting planning scene monitor
[ INFO] [1640013965.427060119]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1640013965.892332548]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1640013965.894481227]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1640013965.904558469]: Constructing new MoveGroup connection for group 'hand' in namespace ''
[ INFO] [1640013967.048063098]: Ready to take commands for planning group hand.

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2021-12-20:
Typically such display issues are due to the OpenGL in use. Please, could you provide more details about your specific setup?
Can you get any standard OpenGL output or generic output in rviz?

mashuai191, 2021-12-20:
Thanks for quick response! Everything looks ok (there is a empty grid chart) except the robot not seen. 
For setup I copied some logs there, hope that useful.

rhaschke, 2021-12-21:
The logs don't indicate an issue. I suspect issues with WSL2 and OpenGL.

mashuai191, 2021-12-23:
@rhaschke, today i fresh-installed a VM(Ubuntu20.04) in VirtualBox and go through the same steps, and the robot can be shown properly, so as you said that may be an issue in my WSL2 and OpenGL. For now i can use my VirtualBox VM to go forward. Thanks for your help!

closed 2021-12-23 by rhaschke

IsolatedMeasure, 2022-02-06:
I run into this problem either, it was the same as the issue in [wslg issue 554](https://github.com/microsoft/wslg/issues/554) ,  seems that ogre3d  in wslg can't work very well. Like the advice given in that issue, by temporary set `export LIBGL_ALWAYS_SOFTWARE=1` , the problem can be solved temporary. Maybe this information is a little bit helpful.