moveit/moveit #3010: Problems in the generation of the IKFast of a custom model

juliamp22, 2021-12-23:
### Description

Hello,
I am trying to use IKFast plugin to test it with a custom robot, I am kindly new so what Iam doing is follow [this](https://ros-planning.github.io/moveit_tutorials/doc/ikfast/ikfast_tutorial.html) tutorial.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? Source
* If binary, which release version? -
* If source, which branch? master branch

### Steps to reproduce
Follow the same steps as in the tutorial:

1. Install the OpenRave version using [this](http://robots.uc3m.es/installation-guides/install-openrave.html) tutorial.
1.1 OptIon A: Install OpenRAVE via scripts (first part in the tutorial): Installs OpenRAVE 0.53 version (it gives me lots of warnings and it seems that there is some incompatibility with python 2.7, but it compiles without any error).

1.2. Option B: Install OpenRAVE 0.85.0 version (second part in the tutorial)
2. Install moveit kinematics: **sudo apt-get install ros-${ROS_DISTRO}-moveit-kinematics**
3. Clone your moveit repository: **git clone https://github.com/ros-planning/moveit.git**
4. Export my robot name: **export MYROBOT_NAME="panda"**
5. Convert the urdf to collada: **rosrun xacro xacro -o $MYROBOT_NAME.urdf $MYROBOT_NAME.urdf.xacro**
6. Generate the IKFast MoveIt plugin: **rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype Transform6D $MYROBOT_NAME.urdf <planning_group_name> <base_link> <eef_link>**

**First execution with Transform 6D:**

### Expected behaviour
After executing the command to generate the IKFast Moveit plugin: **rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype Transform6D $MYROBOT_NAME.urdf <planning_group_name> <base_link> <eef_link>** 

I should obtain an Transform6D IK from the custom robot

### Actual behaviour
It does not created as my robot does not have 6 joints defined.

### Backtrace or Console output

The console gives the following output:

```
Document successfully written to /tmp/ikfast.QLX8t4/robot.dae
Building docker image
Sending build context to Docker daemon  54.63MB
Step 1/3 : FROM personalrobotics/ros-openrave
 ---> 3470beb7a703
Step 2/3 : RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654 &&     apt-key del 421C365BD9FF1F717815A3895523BAEEB01FA116 &&     apt-get update &&     apt-get install -y --no-install-recommends python-pip build-essential liblapack-dev ros-indigo-collada-urdf &&     apt-get clean && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 906eaa471d0b
Step 3/3 : RUN pip install git+https://github.com/sympy/sympy.git@sympy-0.7.1
 ---> Using cache
 ---> f107fec04c69
Successfully built f107fec04c69
Successfully tagged fixed-openrave:latest
Successfully built docker image.
Running openrave0.9.py --database inversekinematics --robot=/input/wrapper.xml --iktype=Transform6D --iktests=1000
could not import scipy.optimize.leastsq
2021-12-23 09:47:26,836 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name fcl_, interface collisionchecker
Traceback (most recent call last):
  File "/usr/bin/openrave0.9.py", line 126, in <module>
    database.run(args=args)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1123, in run
    InverseKinematicsModel.RunFromParser(*args,**kwargs)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1088, in RunFromParser
    model = DatabaseGenerator.RunFromParser(Model=Model,parser=parser,robotatts=robotatts,args=args,**kwargs)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/__init__.py", line 262, in RunFromParser
    model.autogenerate(options=options)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 653, in autogenerate
    self.generate(iktype=iktype,freejoints=freejoints,precision=precision,forceikbuild=forceikbuild,outputlang=outputlang,ipython=ipython,ikfastmaxcasedepth=ikfastmaxcasedepth)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 837, in generate
    raise InverseKinematicsError(u'Manipulator %(manip)s (indices=%(manipindices)r) joint indices to solve for (%(solveindices)r) is not equal to number of expected joints (%(dofexpected)d) for IK type %(iktype)s'%{'manip':self.manip.GetName(),'manipindices':list(self.manip.GetArmIndices()), 'solveindices':list(self.solveindices), 'dofexpected':dofexpected, 'iktype':self.iktype.name})
[... 59 lines cut ...]
                              ^
Traceback (most recent call last):
  File "/usr/bin/openrave0.9.py", line 126, in <module>
    database.run(args=args)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1123, in run
    InverseKinematicsModel.RunFromParser(*args,**kwargs)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1088, in RunFromParser
    model = DatabaseGenerator.RunFromParser(Model=Model,parser=parser,robotatts=robotatts,args=args,**kwargs)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/__init__.py", line 262, in RunFromParser
    model.autogenerate(options=options)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 653, in autogenerate
    self.generate(iktype=iktype,freejoints=freejoints,precision=precision,forceikbuild=forceikbuild,outputlang=outputlang,ipython=ipython,ikfastmaxcasedepth=ikfastmaxcasedepth)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 918, in generate
    objectfiles = compiler.compile(sources=[platformsourcefilename],macros=[('IKFAST_CLIBRARY',1),('IKFAST_NO_MAIN',1)],extra_postargs=compile_flags,output_dir=output_dir)
  File "/usr/lib/python2.7/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/usr/lib/python2.7/distutils/unixccompiler.py", line 122, in _compile
    raise CompileError, msg
distutils.errors.CompileError: command 'cc' failed with exit status 1
docker run --rm --user 1000:1000 -v /tmp/ikfast.S9Mdzx:/input --workdir /input -e HOME=/input fixed-openrave:latest openrave0.9.py --database inversekinematics --robot=/input/wrapper.xml --iktype=TranslationXY2D --iktests=1000\nfailed with exec code 1:
cat: /tmp/ikfast.S9Mdzx/ifast.log: No such file or directory

Thanks for your time,

Júlia

rhaschke, 2021-12-23:
As stated in the [MoveIt IKFast tutorial](https://ros-planning.github.io/moveit_tutorials/doc/ikfast/ikfast_tutorial.html), there is no need to manually install OpenRAVE:
> OpenRAVE is a planning framework as complex as MoveIt itself and installing it is tricky – particularly because its public documentation is not maintained anymore. Fortunately, personalrobotics provide a docker image based on Ubuntu 14.04 with OpenRAVE 0.9.0 and ROS Indigo installed, which can be used to generate the solver code once.

Actually, as you `rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh`, your manually installed OpenRAVE isn't actually used. By the way, a version 0.54 of OpenRAVE never existed. 

Inspecting your error log, it looks like OpenRAVE / IKFast itself fails to compile its self-generated code. There is no easy solution to this as OpenRAVE isn't developed anymore. You can try reducing accuracy as explained in the [_tweaking the process_ section](https://ros-planning.github.io/moveit_tutorials/doc/ikfast/ikfast_tutorial.html#tweaking-the-creation-process) of the tutorial. If that doesn't help, we cannot provide any further support.
To ensure that OpenRAVE / IKFast works in general, you should try for example on the Panda robot.
What robot you are trying to apply IKFast to?

juliamp22, 2021-12-27:
Hello @rhaschke,
Many thanks for your help.

0. As you said I am using the docker image: 

- First, I download it and made a docker run personalrobotics/ros-openrave
- Second I made a docker ps (but no docker container seems to be running)

1. I've tried to generate IKFast from panda_moveit_config [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config) (noetic-devel)

- rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype Transform6D panda.urdf panda_arm panda_link0 panda_link8

It seems that the IK it is well generated but it ends with the following error:

`openravepy.databases.inversekinematics: testik, success rate: 1.000000, wrong solutions: 0.000000, no solutions: 0.000000, missing solution: 0.171000
Created /tmp/ikfast.wyUUdA/.openrave/kinematics.ecd9cbb23ff18ce9438f30f09eede2b8/ikfast0x10000049.Transform6D.0_1_2_3_4_5_f6.cpp

Running /home/julia/ws_moveit/src/moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts/create_ikfast_moveit_plugin.py "panda" "panda_arm" "panda_panda_arm_ikfast_plugin" "panda_link0" "panda_link8" "/tmp/ikfast.wyUUdA/.openrave/kinematics.ecd9cbb23ff18ce9438f30f09eede2b8/ikfast0x10000049.Transform6D.0_1_2_3_4_5_f6.cpp"
Traceback (most recent call last):
  File "/home/julia/ws_moveit/src/moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts/create_ikfast_moveit_plugin.py", line 51, in <module>
    import yaml
ImportError: No module named yaml
`

2. I've tried to _tweaking the process section_ as you said, and it gives the following error:
rosrun moveit_kinematics round_collada_numbers.py panda.dae panda.rounded.dae 5

- `Collada Number Rounder
Rounding numbers to 5 decimal places

Traceback (most recent call last):
  File "/home/julia/ws_moveit/devel/lib/moveit_kinematics/round_collada_numbers.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/julia/ws_moveit/src/moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts/round_collada_numbers.py", line 97, in <module>
    dom = etree.parse(io.BytesIO(xml))
TypeError: a bytes-like object is required, not 'str'
`

Regards and thanks,

Júlia

simonschmeisser, 2021-12-27:
Running the docker is handled inside this script: https://github.com/ros-planning/moveit/blob/master/moveit_kinematics/ikfast_kinematics_plugin/scripts/auto_create_ikfast_moveit_plugin.sh  and the containers will be destroyed immediately, it's normal to see no trace of them in `docker ps`

Do you have `python3-yaml` installed?

The error in `round_collade_numbers was fixed recently, please update your checkout (`git pull`)  #2983

juliamp22, 2021-12-27:
Hello,
Thanks for all,
I just have installed:

`sudo apt-get install python-yaml
sudo apt-get install python-lxml`

It seems that is working now:

```
Running /home/julia/ws_moveit/src/moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts/create_ikfast_moveit_plugin.py "panda" "panda_arm" "panda_panda_arm_ikfast_plugin" "panda_link0" "panda_link8" "/tmp/ikfast.4hVLR8/.openrave/kinematics.ecd9cbb23ff18ce9438f30f09eede2b8/ikfast0x10000049.Transform6D.0_1_2_3_4_5_f6.cpp"
Failed to import roslib. No ROS environment available? Trying without.
Creating IKFastKinematicsPlugin with parameters: 
 robot_name:           panda
 base_link_name:       panda_link0
 eef_link_name:        panda_link8
 planning_group_name:  panda_arm
 ikfast_plugin_pkg:    panda_panda_arm_ikfast_plugin
 ikfast_output_path:   /tmp/ikfast.4hVLR8/.openrave/kinematics.ecd9cbb23ff18ce9438f30f09eede2b8/ikfast0x10000049.Transform6D.0_1_2_3_4_5_f6.cpp
 search_mode:          OPTIMIZE_MAX_JOINT
 srdf_filename:        panda.srdf
 robot_name_in_srdf:   panda
 moveit_config_pkg:    panda_moveit_config

Found source code generated by IKFast version 73
Failed to find package: panda_panda_arm_ikfast_plugin. Will create it in /home/julia/ws_moveit/src/panda_panda_arm_ikfast_plugin.
Created package.xml at: '/home/julia/ws_moveit/src/panda_panda_arm_ikfast_plugin/package.xml'
Created ikfast header file at '/home/julia/ws_moveit/src/panda_panda_arm_ikfast_plugin/include/ikfast.h'
Created ikfast plugin file at '/home/julia/ws_moveit/src/panda_panda_arm_ikfast_plugin/src/panda_panda_arm_ikfast_moveit_plugin.cpp'
Created plugin definition at  '/home/julia/ws_moveit/src/panda_panda_arm_ikfast_plugin/panda_panda_arm_moveit_ikfast_plugin_description.xml'
Created cmake file at '/home/julia/ws_moveit/src/panda_panda_arm_ikfast_plugin/CMakeLists.txt'
Wrote package.xml at  '/home/julia/ws_moveit/src/panda_panda_arm_ikfast_plugin/package.xml'
Created update plugin script at '/home/julia/ws_moveit/src/panda_panda_arm_ikfast_plugin/update_ikfast_plugin.sh'
Failed to update MoveIt package:
Failed to find package: panda_moveit_config

```
It seems that the package can't found _moveit_ package nor _panda_moveit_config_ but it seems a problem in my ROS installation, I will look for it.

Thanks a lot for you time,

Júlia

rhaschke, 2021-12-28:
ROS packages were not found, because you didn't source your ROS environment.
Also, you are using python2 as your default python interpreter in Ubuntu 20.04, which is discouraged. You should switch to python3 as the default (or not use a default at all, but explicitly run python2 vs. python3).
https://askubuntu.com/questions/1296790/python-is-python3-package-in-ubuntu-20-04-what-is-it-and-what-does-it-actually

If your issue is solved, please close this issue. Thanks.

juliamp22, 2021-12-28:
Hello @rhaschke,
Many thanks for your response. 
- Yes, I did source my ROS environment but it still does not find the franka_description package, but if i make the rospack find of the package, it shows where it is, strange things.
- I also change the default python version to python 3.
- In my custom .urdf model now I could obtain the .dae tweaking it but, changing the code: `dom = etree.parse(input_file)` to 
 ```
   p = etree.XMLParser(huge_tree=True)
    dom = etree.parse(input_file,parser=p)
```
- I still trying to obtain the ikfast_moveit_plugin using the TranslationXY2D from my custom model but, it still appears the following:

```
openravepy.databases.inversekinematics: generate, compiling ik file to /input/.openrave/kinematics.fc72b194e423dc404401398e444ba8b1/ikfast0x10000049.TranslationXY2D.x86_64.0_1_f2.so
/input/.openrave/kinematics.fc72b194e423dc404401398e444ba8b1/ikfast0x10000049.TranslationXY2D.x86_64.0_1_f2.cpp: In member function ‘bool IKSolver::ComputeIk(const IkReal*, const IkReal*, const IkReal*, ikfast::IkSolutionListBase<double>&)’:
/input/.openrave/kinematics.fc72b194e423dc404401398e444ba8b1/ikfast0x10000049.TranslationXY2D.x86_64.0_1_f2.cpp:362:30: error: ‘pz’ was not declared in this scope
 j0array[0]=((7.000000035e-8)+pz+(((0.0001)*px)));
```

- What I've done also is test the same with the panda robot but it gives me the following message:

rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype TranslationXY2D panda.urdf panda_arm panda_link0 panda_link8

RuntimeError: maximum recursion depth exceeded
docker run --rm --user 1000:1000 -v /tmp/ikfast.EMJ2ip:/input --workdir /input -e HOME=/input fixed-openrave:latest openrave0.9.py --database inversekinematics --robot=/input/wrapper.xml --iktype=TranslationXY2D --iktests=1000\nfailed with exec code 1:
cat: /tmp/ikfast.EMJ2ip/ifast.log: No such file or directory

Thanks for all,

Júlia

rhaschke, 2021-12-28:
I [already wrote](https://github.com/ros-planning/moveit/issues/3010#issuecomment-1000223039) that your particular robot kinematics seems to break IKFast. There is nothing we can do about it as OpenRave is an external package. If tweaking the precision didn't help, this is a dead end.

> In my custom .urdf model, now I could obtain the .dae tweaking, but changing the code: `dom = etree.parse(input_file)`

I don't see why you need the mentioned code change. That's not needed on our side.

I just noticed that you are attempting to use `TranslationXY2D`. This is probably much less tested in OpenRAVE (and on our side).

rhaschke, 2021-12-28:
> `RuntimeError: maximum recursion depth exceeded`

I was able to reproduce this error, using `TranslationXY2D` for the panda robot. It originates from OpenRAVE.
Unfortunately, we cannot do anything about it in this case. I'm sorry that we cannot help in your case.
I suggest manually finding the inverse kinematics for the two-dimensional plane, which is typically rather simple using some trigonometry. This would be much faster as well than IKFast.

closed 2021-12-28 by rhaschke

juliamp22, 2021-12-29:
Hello @rhaschke,
Many thanks for your help.
Yes, I will continue getting the IK manually from the two-dimensional plate and also review again the custom urdf.

Best Regards,

Júlia

ruiaisun, 2022-02-15:
Hello, @juliamp22 @rhaschke 
I am also trying to use OpenRAVE ikfast to test it with panda and ur5 robot. Do @juliamp22 you finally find the solution for the iktype 'TranslationXY2D' ? For me, I have generated successfully the ikfast plugin for both robot with iktype is 'Transform6D' by using inversekinematics Module here (http://openrave.org/docs/latest_stable/openravepy/databases.inversekinematics/). However, it does not work for the 'Translation3D'. Do you guys have any suggestions for that?

juliamp22, 2022-02-16:
Hello @ruiaisun,
I tried to obtain TranslationXY2D with panda and also Translation3D without success.
As far as I know, openRAVE are not going to be mantained anymore, so, maybe what you could do is use the same docker image as it is used in moveit noetic explanation to obtain the IK with openrave:
https://ros-planning.github.io/moveit_tutorials/doc/ikfast/ikfast_tutorial.html

After multiple attempts with openrave and without having success, I tried the TRACK-IK algorithm, it worked without any problem. I highly recommend that if it may fit you:

https://ros-planning.github.io/moveit_tutorials/doc/trac_ik/trac_ik_tutorial.html

Hope that may help you,

Júlia