moveit/moveit #3018: Select the free index during the IKFast plugin generation

fstoriale, 2022-01-04:
### Description

Hello everyone,

I was trying to create an IKFast kinematic solver plugin for a 7 DOF robot in ROS Noetic. Following the MoveIt! tutorial I was able to generate the plugin but it seems it is not possible to choose the free index as before. Indeed, in the old tutorial for ROS Melodic and Kinetic it was required to specify the free index, i.e. the redundant joint, directly using the docker image from personalrobotics. Now it is chosen automatically by the plugin generator but I need to choose it by myself. Is it possible?

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04

### Steps to reproduce
This is the suggested instruction in the MoveIt! tutorial:
`rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype Transform6D $MYROBOT_NAME.urdf <planning_group_name> <base_link> <eef_link>`

### Expected behaviour
I would expect to have the possibility to choose the free index. Indeed, the command to generate the solver through docker is 
`python openrave-config --python-dir /openravepy/_openravepy_/ikfast.py --robot="$MYROBOT_NAME".dae --iktype=transform6d --baselink="$BASE_LINK" --eelink="$EEF_LINK" --freeindex="$FREE_INDEX" --savefile="$IKFAST_OUTPUT_PATH"`

### Actual behaviour
Using the auto_create_ikfast_moveit_plugin.sh the free index is automatically chosen

rhaschke, 2022-01-04:
We don't call the ikfast.py script directly anymore, but use the recommended OpenRAVE wrapper script:
https://github.com/ros-planning/moveit/blob/8770a18f752e5e7c21fd0e029af3fd2990b8ab67/moveit_kinematics/ikfast_kinematics_plugin/scripts/auto_create_ikfast_moveit_plugin.sh#L167

To pass a custom free joint, you could perform the steps of `auto_create_ikfast_moveit_plugin.sh` manually and pass the desired argument(s):
```
> openrave0.9.py --database inversekinematics -h
could not import scipy.optimize.leastsq
Usage: openrave.py --database inversekinematics [options]

Uses ikfast to compute the closed-form inverse kinematics equations of a robot
manipulator, generates a C++ file, and compiles this file into a shared object
which can then be loaded by OpenRAVE.

Options:
  -h, --help            show this help message and exit
  --freejoint=FREEJOINTS
                        Optional joint name specifying a free parameter of the
                        manipulator. The value of a free joint is known at
                        runtime, but not known at IK generation time. If
                        nothing specified, assumes all joints not solving for
                        are free parameters. Can be specified multiple times
                        for multiple free parameters.
  --precision=PRECISION
                        The precision to compute the inverse kinematics in,
                        (default=8).
  --maxcasedepth=MAXCASEDEPTH
                        The max depth to go into degenerate cases. If ikfast
                        file is too big, try reducing this, (default=3).
  --usecached           If set, will always try to use the cached ik c++ file,
                        instead of generating a new one.
  --freeinc=FREEINC     The discretization value of freejoints.
  --numiktests=IKTESTS, --iktests=IKTESTS
                        Will test the ik solver and return the success rate.
                        IKTESTS can be an integer to specify number of random
                        tests, it can also be a filename to specify the joint
                        values of the manipulator to test. The formst of the
                        filename is #numiktests [dof values]*
  --iktestjthresh=IKTESTJTHRESH
                        When testing configurations, the eigenvalues of the
                        jacobian all have to be greater than this value
  --perftiming=PERFTIMING
                        Number of IK calls for measuring the internal ikfast
                        solver.
  --outputlang=OUTPUTLANG
                        If specified, will output the generated code in that
                        language (ie --outputlang=cpp).
  -i, --ipython         if true will drop into the ipython interpreter right
                        before ikfast is called
  --iktype=IKTYPE       The ik type to build the solver current types are:
                        Transform6D, Rotation3D, Translation3D, Direction3D,
                        Ray4D, Lookat3D, TranslationDirection5D,
                        TranslationXY2D, TranslationXYOrientation3D,
                        TranslationLocalGlobal6D, TranslationXAxisAngle4D,
                        TranslationYAxisAngle4D, TranslationZAxisAngle4D,
                        TranslationXAxisAngleZNorm4D,
                        TranslationYAxisAngleXNorm4D,
                        TranslationZAxisAngleYNorm4D, CustomDataBit
```

rhaschke, 2022-01-04:
Ideally, [`auto_create_ikfast_moveit_plugin.sh`](https://github.com/ros-planning/moveit/blob/8770a18f752e5e7c21fd0e029af3fd2990b8ab67/moveit_kinematics/ikfast_kinematics_plugin/scripts/auto_create_ikfast_moveit_plugin.sh) is augmented to forward unhandled arguments to OpenRAVE.
A corresponding PR is very welcome.