moveit/moveit #3028: Proposal to Include BeBOT Motion Planning Framework

magicbycalvin, 2022-01-17:
**Proposal**

I would like to propose the inclusion of the BeBOT motion planning framework into MoveIt. BeBOT, which stands for Bernstein/Bézier Optimal Trajectories, leverages Bernstein polynomials to solve optimal trajectory generation problems. Bernstein polynomials provide convenient geometric properties that enable the trajectory planner to efficiently evaluate and impose constraints along the vehicles' trajectories, such as maximum speed and angular rates, minimum distance between trajectories and between the vehicles and obstacles. These safety and feasibility constraints can be guaranteed in continuous time and may significantly reduce the dimensionality of motion planning problems.

**Resources**

Relevant links discussing the method in more detail can be found below:
* Overview presentation of BeBOT for the Reddit Robotics Symposium: https://youtu.be/QR50lx9ylKE?t=6103
* GitHub repo: https://github.com/caslabuiowa/BeBOT
* Original conference paper for IROS 2019: https://ieeexplore.ieee.org/document/8967564

_Note that the project on GitHub is written in Python and does not meet my own standards for a deployable package yet._

**Future Work**

There is currently a paper under review which significantly expands on the original conference paper. There, more properties are included along with properties of rational Bernstein polynomials. It also includes additional examples such as generating trajectories for 1000 aerial vehicles and generating trajectories for a Traveling Salesman-type mission.

I am currently investigating the use of BeBOT with uncertain systems such as partially known vehicle dynamics, search and rescue for a victim whose beacon has considerable variance, and avoiding stochastic obstacles. Due to the structure of my method, BeBOT tends to be very flexible and should lend itself well to unforeseen future development.

**Additional Information**

I am a PhD student and my dissertation will cover BeBOT along with related extensions, so you can expect that I will continue to develop this work at least until I graduate (however as of right now I do plan on developing and supporting this work indefinitely). Other than some small changes here and there, I have not significantly contributed to an open source project before but I am willing and able to put the work in to ensure that I make helpful contributions.

Through the Discord server, @JafarAbdi is the original person that suggested I open this issue. They also mentioned that this could be a good candidate for a GSOC project.

Thank you very much for your time and consideration!

gavanderhoorn, 2022-01-17:
This sounds interesting, especially the integrated support for constraints and the continuous nature.

Two questions:

 1. you mention 'vehicles' quite a few times. MoveIt is primarily a planning framework for manipulators. The math is likely generic enough, but is BeBOT similarly generic to be useful in an environment which focuses on planning motions for (robotic) manipulators?
 2. would you have any idea already how extensively MoveIt would need to be changed to support this integration? How well do the domain concepts map between MoveIt and BeBOT? Would this be something which could be implemented out-of-tree, or would significant infrastructural changes also be needed?

magicbycalvin, 2022-01-17:
1. You are correct about "vehicles". The original design was built for mobile robots. Fortunately though, BeBOT is sufficiently general that it can be applied to manipulators. I have not yet written an example covering this, but other than a couple subtleties I'll have to learn when it comes to algorithms for manipulators, it shouldn't be a problem. In addition to that, I believe that it would make a very nice interface for driving mobile manipulators or even generating hybrid motion/manipulation plans. In general, if the problem can be formulated as a nonlinear optimization problem, BeBOT can be partially or fully utilized (partial utilization in this context means using Bernstein polynomials for most cost/constraint functions but also including some cost or constraint that might be better represented using a different method).
2. In my excitement, I overlooked this problem. I do not sufficiently understand the MoveIt framework to give you a good answer to this question yet. I will look into this and get back to you. If you have any resources you can share that will hasten my research regarding this problem, I would appreciate it!
3. You are correct that the original implementation is written in Python. It is hard to say for sure whether writing the library in C++ or using a ROS API would be the better approach. I am more than capable of either route, although I am more experienced with Python. If you have any suggestions as to which route would be preferable, I would love to hear them. In the meantime, I will look into this as well and get back to you with a solid answer.

Thanks!

rhaschke, 2022-01-18:
This looks interesting definitely. Given the capabilities of BeBOT I consider it a candidate for a [`PlannerManager` plugin](https://github.com/ros-planning/moveit/blob/802e596a2283b64f4582d802c5f79e1f3d57def0/moveit_core/planning_interface/include/moveit/planning_interface/planning_interface.h#L150). A C++ implementation would be most suitable in my opinion. However, doesn't that mean reimplementing everything, given that your current implementation is Python / MATLAB?

magicbycalvin, 2022-01-18:
@rhaschke My current implementation is written in Python. However, if a C++ implementation would be more suitable, that shouldn't be a problem. The majority of the work has been in the research and development of the method itself rather that writing the code. Because of that, I foresee a C++ implementation not taking much more time than modifying the existing Python code to fit the format for MoveIt.

In addition to the topics I already mentioned I will be investigating above, I'll take a look at the plugin that you linked and start coming up with a game plan for this work.

Thanks!

rhaschke, 2022-01-19:
I'm looking forward to your contribution! Thanks in advance.