moveit/moveit #3034: --- stderr: moveit_ros_planning

Sting2021, 2022-01-22:
### Description

Hi everyone. 
I'm trying to install Moveit2 with my ROS 2 Galactic environment but i'm having this issue when configuring my colcon workspace.
How can I resolve, please?

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Binary build
* ROS 2

### Steps to reproduce
Building colcon workspace after installing step by step Moveit 2 with GETTING STARTED guide 

### Expected behaviour
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Finished <<< moveit_common [0.64s]                                        
Starting >>> launch_param_builder
Finished <<< moveit_resources_panda_description [0.69s]                   
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_panda_moveit_config [0.84s]                  
Starting >>> moveit_resources_fanuc_description
Finished <<< geometric_shapes [1.62s]                                   
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_fanuc_description [0.44s]               
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_pr2_description [0.42s]
Starting >>> moveit_resources_prbt_support
Finished <<< moveit_resources_fanuc_moveit_config [0.31s]             
Starting >>> rviz_visual_tools
Finished <<< moveit_resources_prbt_support [0.34s]
Starting >>> moveit_resources
Finished <<< moveit_resources [0.46s]                                 
Finished <<< launch_param_builder [2.29s]                             
Starting >>> moveit_configs_utils
Finished <<< rviz_visual_tools [1.58s]                                 
Finished <<< moveit_configs_utils [1.75s]                              
Finished <<< moveit_msgs [7.29s]                       
Starting >>> moveit_core
Finished <<< moveit_core [2.04s]                       
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> chomp_motion_planner
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [0.57s]
Starting >>> pilz_industrial_motion_planner_testutils
Finished <<< moveit_simple_controller_manager [0.66s]
Starting >>> moveit_plugins
Finished <<< chomp_motion_planner [0.70s]
Starting >>> moveit_chomp_optimizer_adapter
Finished <<< moveit_ros_occupancy_map_monitor [0.80s]
Starting >>> moveit_ros_planning
Finished <<< moveit_plugins [0.47s]     
Starting >>> moveit_planners_chomp
Finished <<< pilz_industrial_motion_planner_testutils [0.94s]
Starting >>> moveit_ros_control_interface
Finished <<< moveit_chomp_optimizer_adapter [0.84s]
Finished <<< moveit_planners_chomp [0.88s]                                     
Finished <<< moveit_ros_control_interface [0.60s]

{...}

### Backtrace or Console output
--- stderr: moveit_ros_planning                                 
/home/federico/ws_moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp: In member function ‘void planning_scene_monitor::PlanningSceneMonitor::excludeWorldObjectFromOctree(const ObjectConstPtr&)’:
/home/federico/ws_moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:945:81: error: ‘using element_type = const struct collision_detection::World::Object’ {aka ‘const struct collision_detection::World::Object’} has no member named ‘global_shape_poses_’; did you mean ‘shape_poses_’?
  945 | y_shape_handles_[obj->id_].push_back(std::make_pair(h, &obj->global_shape_poses_[i]));
      |                                                              ^~~~~~~~~~~~~~~~~~~
      |                                                              shape_poses_
/home/federico/ws_moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp: In member function ‘bool planning_scene_monitor::PlanningSceneMonitor::getShapeTransformCache(const string&, const rclcpp::Time&, occupancy_map_monitor::ShapeTransformCache&) const’:
/home/federico/ws_moveit2/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:1137:47: error: ‘const class moveit::core::AttachedBody’ has no member named ‘getShapePosesInLinkFrame’
 1137 |             attached_body_shape_handle.first->getShapePosesInLinkFrame()[attached_body_shape_handle.second[k].second];
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/build.make:63: planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/src/planning_scene_monitor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:876: planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_ros_planning [25.0s, exited with code 2]

Summary: 23 packages finished [35.5s]
  1 package failed: moveit_ros_planning
  1 package had stderr output: moveit_ros_planning
  21 packages not processed

Thank you!

JafarAbdi, 2022-01-25:
I think the problem is you have moveit install from both source/debians, running `sudo apt remove --purge ros-galactic-moveit*` before compiling should fix it, I'm closing this since it's a https://github.com/ros-planning/moveit2/ issue, please create an issue there if the proposed solution still not works

closed 2022-01-25 by JafarAbdi

zouzhe1, 2022-08-12:
@JafarAbdi  i have this problem too.     ubuntu22.04  ros2   moveit2