moveit/moveit #3035: Pilz industrial motion planner ignores goal constraint frame_id

nicola-sysdesign, 2022-01-25:
### Description

Hi, I'm using pilz_industrial_motion_planner to plan some trajectories interfacing directly with it through some calls to /move_group action server.

I noticed that if I plan with OMPL pipeline, I can set the goal constraint frame_id as any frame present in the planning scene (TF, collision objects sub-frames) and the planner correctly plan to the right goal as desired.

Instead planning with pilz_industrial_motion_planner pipeline:
* PTP simply ignores the goal constraint frame_id and instead it assumes that goal position and orientation constraints are respect to the robot model frame.
* LIN always fails reporting: 
```
[ INFO] [1643103242.859113763]: Using planning pipeline 'pilz_industrial_motion_planner'
[ INFO] [1643103242.859239956]: Generating LIN trajectory...
[ERROR] [1643103242.859357051]: Given frame (goal_link) is unequal to model frame(world)
[ERROR] [1643103242.859459598]: Failed to compute inverse kinematics for link: tool0 of goal pose
```
* CIRC I've not tested yet but I suppose there is some issue there too.

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Binary build
* 1.1.7

### Steps to reproduce
* Create a MotionPlanRequest with a goal constraint frame_id equal to a TF frame or collision object sub-frame present in the planning scene;
* Send a goal with that motion request to /move_group action server.

### Expected behaviour
Plan taking account of the goal constraint frame_id in the planning request as does OMPL.

### Actual behaviour
* PTP ignores the goal constraint frame_id.
* LIN always fails.
* CIRC not yet tested.

rhaschke, 2022-01-25:
It would be great if you could provide a minimal example (best in the form of a failing unit test) to reproduce the issue, e.g. with [MoveIt's Panda robot config](https://github.com/ros-planning/moveit_resources/tree/master/panda_moveit_config).

nicola-sysdesign, 2022-01-26:
I managed to write a small node example that show the issue.
Launch Panda MoveIt Demo: `roslaunch moveit_resources_panda_moveit_config demo.launch`

Than run the node below:
```
#include <string>
#include <vector>
// ROS
#include <ros/ros.h>
#include <ros/console.h>
#include <actionlib/client/simple_action_client.h>
#include <geometry_msgs/Pose.h>
#include <geometry_msgs/PoseStamped.h>
#include <shape_msgs/SolidPrimitive.h>
// MoveIt
#include <moveit/kinematic_constraints/utils.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit_msgs/CollisionObject.h>
#include <moveit_msgs/Constraints.h>
#include <moveit_msgs/ApplyPlanningScene.h>
#include <moveit_msgs/MoveGroupAction.h>

const std::string ACTION_NAME("/move_group");

const std::string PIPELINE_ID("pilz_industrial_motion_planner");
const std::string PLANNER_ID("PTP");
// const std::string PIPELINE_ID("ompl");         // UNCOMMENT THIS LINES TO SEE THAT OMPL WORKS
// const std::string PLANNER_ID("RRTConnect");    // UNCOMMENT THIS LINES TO SEE THAT OMPL WORKS

const std::string GROUP_NAME("panda_arm");
const std::string LINK_NAME("panda_hand");

int main(int argc, char* argv[])
{
  ros::init(argc, argv, "move_over_the_box_node");

  // Node
  ros::NodeHandle node("~");

  ros::AsyncSpinner spinner(2);
  spinner.start();

  // Action Client
  actionlib::SimpleActionClient<moveit_msgs::MoveGroupAction> move_group_acli(node, ACTION_NAME);

  /* PLANNING SCENE INTERFACE */
  moveit::planning_interface::PlanningSceneInterface planning_scene_interface;

  // add object to the planning scene
  moveit_msgs::CollisionObject collision_object;
  collision_object.header.frame_id = "world";
  collision_object.id = "box";
  collision_object.pose.position.x = 0.5;
  collision_object.pose.position.z = 0.1;
  collision_object.pose.orientation.w = 1.;
  // add box shape
  {
    shape_msgs::SolidPrimitive primitive;
    primitive.type = shape_msgs::SolidPrimitive::BOX;
    primitive.dimensions.resize(3);
    primitive.dimensions[shape_msgs::SolidPrimitive::BOX_X] = 0.2;
    primitive.dimensions[shape_msgs::SolidPrimitive::BOX_Y] = 0.2;
    primitive.dimensions[shape_msgs::SolidPrimitive::BOX_Z] = 0.2;
    collision_object.primitives.push_back(primitive);
    geometry_msgs::Pose pose;
    pose.orientation.w = 1.;
    collision_object.primitive_poses.push_back(pose);
  }
  planning_scene_interface.applyCollisionObject(collision_object);

  // create a motion plan request
  moveit_msgs::MoveGroupGoal goal;
  goal.request.pipeline_id = PIPELINE_ID;
  goal.request.planner_id = PLANNER_ID;
  goal.request.group_name = GROUP_NAME;
  goal.request.max_velocity_scaling_factor = 1.0;
  goal.request.max_acceleration_scaling_factor = 1.0;
  goal.request.allowed_planning_time = 5.0;
  {
    geometry_msgs::PoseStamped goal_pose;
    goal_pose.header.frame_id = "box";
    goal_pose.pose.position.z = 0.6;
    goal_pose.pose.orientation.x = 1.;

    moveit_msgs::Constraints goal_constraints =
        kinematic_constraints::constructGoalConstraints(LINK_NAME, goal_pose);
    goal.request.goal_constraints.push_back(goal_constraints);
  }

  // wait for server
  ros::Duration timeout(5.0);
  if (!move_group_acli.waitForServer(timeout))
  {
    ROS_WARN_STREAM("Waiting for '" << ACTION_NAME << "' Action Server ... timeout.");
    return 0;
  }

  // send goal and wait for result
  move_group_acli.sendGoalAndWait(goal);
  moveit_msgs::MoveGroupResult::ConstPtr result = move_group_acli.getResult();

  // ros::waitForShutdown();
  return 0;
}
```
As you will see the robot will move above its own base instead of above the cube.

anuj018, 2022-08-11:
Any update on if you were able to set the frame id for your trajectories?

rhaschke, 2023-11-10:
I think, one culprit here is the missing `ResolveConstraintFrames` planning request adapter for the Pilz pipeline.
This causes the goal frame to be not resolvable for the planner.

PR #3522 (2023-11-11): Tranform Pilz goal pose and center point wrt. the planning frame
### Description

Should fix #3035 (partially) and ros-planning/moveit_task_constructor/issues/411 by transforming the goal pose (CIRC, LIN and PTP) and the center point (CIRC) wrt. the planning frame.

Only computes the IK when there is a goal pose constraint. Removed unused `frame_id` initialisation. Transforms the goal pose for `CIRC`, `LIN` and `PTP` wrt. the planning frame. Transform the center point for `CIRC` wrt. the planning frame.

Rebased on #3519.

**TODO**
- [x] Fix CIRC
- [x] Fix LIN and PTP
- [ ] Add unit tests
- [  [... cut]
closed 2023-11-11 by rhaschke