moveit/moveit #3037: `moveit_ros_benchmarks` MoveIt 1.1.7 segmentation fault

bonedoz, 2022-01-26:
### Description

I followed the [MoveIt benchmarking tutorial for Noetic](https://ros-planning.github.io/moveit_tutorials/doc/benchmarking/benchmarking_tutorial.html) step by step to learn about the benchmarking tool, but I got an error when running the benchmarks by launchin​g `demo_panda.launch containing` the `moveit_run_benchmark node`. I later traced the error using GDB, and the error seems to be caused by a `std::vector access` out of bounds in the call to `moveit_ros_benchmarks::BenchmarkExecutor::runBenchmarks`. For comparison, I followed the [benchmarking tutorial on Melodic](https://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/benchmarking/benchmarking_tutorial.html) (with **MoveIt 1.0.8** installed ) and did the same thing and it worked without problems and benchmarking ran successfully. So is this an error caused by some bug in **MoveIt 1.1.7**?

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04 (kernel 5.11.0-40-generic)
* Binary build
* MoveIt 1.1.7

### Steps to reproduce
`roslaunch moveit_ros_benchmarks demo_panda.launch`

### Expected behaviour
generate benchmark log files

### Actual behaviour
```bash
Thread 1 "moveit_run_benc" received signal SIGSEGV, Segmentation fault.
0x00007ffff7f81674 in std::operator+<char, std::char_traits<char>, std::allocator<char> > (__lhs=__lhs@entry=0x7ffff7fb3e05 "path_", 
    __rhs=<error reading variable: Cannot access memory at address 0x8>)
    at /usr/include/c++/9/bits/char_traits.h:335
```

### Backtrace or Console output
https://gist.github.com/bonedoz/436084dfc89fc7ede4cc20866f4d448e

SantiagoCanete, 2022-05-03:
I am having the same issue in `MoveIt 1.1.9`. I am trying to run the examples from the Noetic tutorials and although the planner is set in the benchmark configuration file `demo1.yaml` to OMPL it defaults to CHOMP.
```
[ INFO] [1651585062.424733103]: Multiple planning plugins available. You should specify the '~planning_plugin' parameter. Using 'chomp_interface/CHOMPPlanner' for now.
[ INFO] [1651585062.436282821]: Using planning interface 'CHOMP'
```
Resulting in:
`[ERROR] [1651585062.645033]: Mongo process exited with error code 48`

rhaschke, 2022-05-03:
This is an issue with your robot-specific moveit_config. Since MoveIt 1.1.2 multiple pipeline configs are supported in parallel, defined as `~planning_pipelines/*`. Check that the parameter /move_group/default_planning_pipeline points to your desired default, like here for `panda_moveit_config`:
```
$ rosparam list | grep planning_plugin
/move_group/planning_pipelines/chomp/planning_plugin
/move_group/planning_pipelines/ompl/planning_plugin
/move_group/planning_pipelines/pilz_industrial_motion_planner/planning_plugin

$ rosparam get /move_group/default_planning_pipeline
ompl
```
@henningkayser: I see more and more people reporting issues with the new config scheme. Did you document the corresponding config changes somewhere, so that we can refer to that?

SantiagoCanete, 2022-05-03:
@rhaschke Thanks! 

yeah, /move_group/default_planning_pipeline points to ompl. 

I am using the `panda_moveit_config noetic-devel` and the config file for the benchmarks is as given in the [examples](https://github.com/ros-planning/moveit/blob/21acae7fbecbca343f28bc912043f64a6a59c13e/moveit_ros/benchmarks/examples/demo1.yaml). I have not made any other configuration changes other than changing the output_directory for the benchmarks in demo1.yaml.

rhaschke, 2022-05-03:
I digged deeper and found that the issue stems from `moveit_ros_benchmark` using a different pipeline config process than `move_group`. Specifically, the namespace `ompl` became redundant in some launch file: https://github.com/ros-planning/moveit/pull/3128.
Please validate the modified launch file.

SantiagoCanete, 2022-05-03:
Yeap, this fixes it for me. Thank you.

closed 2022-05-03 by rhaschke

SebiGoebel, 2023-11-25:
I am having a similar issue. I am tyring to run benchmarks with a ur3e and also getting this message:
```
[ INFO] [1651585062.424733103]: Multiple planning plugins available. You should specify the '~planning_plugin' parameter. Using 'chomp_interface/CHOMPPlanner' for now.
[ INFO] [1651585062.436282821]: Using planning interface 'CHOMP'
```
However when i look at the `planning_plugins` and the `default_planning_pipeline` I get the following:
```
$ rosparam list | grep planning_plugin
/move_group/planning_plugin
$ rosparam get /move_group/default_planning_pipeline
''

``` 

How can I fix this issue? How can I set the `planning_plugin` ?