moveit/moveit #3041: set_path_constraints does not work on compute_cartesian_path

jih189, 2022-01-27:
### Description

I am writing Python code with MoveIt. I am currently using compute_cartesian_path to follow a sequence of way points, but it always easily get too close to the joint limits, so I use set_path_constraints with joint_constraints to avoid the arm getting too close to the joint limits. However, I found the computed path by compute_cartesian_path will still get close to the joint limits and get outside of joint constraints. Then I have test the set_path_constraints with regular plan, and it will return a plan which is correct and not get outside of joint constraints. Therefore, I do some search and found this post 
https://github.com/ros-planning/moveit/issues/20#issuecomment-241679607
, so it looks like the set_path_constraints does not support compute_cartesian_path. Can anyone fix it? Thanks

My environment
ROS Distro: [Melodic]
OS Version: Ubuntu 18.04
Binary build
moveit-commander                1.0.8
moveit-core                     1.0.8
moveit-python                   0.4.1
moveit-ros-planning-interface   1.0.8
moveit-ros-visualization        1.0.8
rqt-moveit                      0.5.10

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2022-01-27:
`compute_cartesian_path` isn't actually planning anything. It's just a linear interpolator in Cartesian space. So, won't fix.

closed 2022-01-27 by rhaschke

jih189, 2022-01-27:
Hi. I have read the moveit source code again, and I found you guys have fix problem already. In https://github.com/ros-planning/moveit/blob/b098c33f4da87c1883bb5a2fbebaf30c186e0fb0/moveit_ros/planning_interface/move_group_interface/src/wrap_python_move_group.cpp#L797, you have a python wrap to handle the constraint, so I you think just need to update the document so people can know that constraints must be passed to compute_cartesian_path.

rhaschke, 2022-01-28:
You are right, you can pass path constraints. However, these are not actively maintained, but only passively validated along the linearly interpolated path.