moveit/moveit #3045: MotionPlanning RViz plugin have to be manually refreshed after clear_octomap

FSund, 2022-01-31:
### Description

After calling the `/clear_octomap` service, the MotionPlanning RViz plugin have to be reloaded (uncheck and re-check the plugin, or remove/add), for the octomap to disappear from the Planning Scene.

The same happens if an **empty** octomap is loaded from a file via `/move_group/load_map` (the old octomap is still shown until the plugin is reloaded). The empty map is generated via `/move_group/save_map` on a fresh scene.

The same behaviour is present when calling the `/apply_planning_scene` service with an empty octomap:
(untested code, it is a minimal version of the actual code I use)
```
get_planning_scene = rospy.ServiceProxy('/get_planning_scene', GetPlanningScene)
components = PlanningSceneComponents()
components.components = sum([
    PlanningSceneComponents.WORLD_OBJECT_NAMES,
    PlanningSceneComponents.WORLD_OBJECT_GEOMETRY,
    PlanningSceneComponents.OCTOMAP
])
ps = get_planning_scene(components)
ps = ps.scene  # moveit_msgs/PlanningScene
ps.is_diff = True
ps.world.octomap = OctomapWithPose()
apply_planning_scene = rospy.ServiceProxy('/apply_planning_scene', ApplyPlanningScene)
apply_planning_scene(ps)
```

I am not sure if any of these 3 methods of clearing the octomap are related, but I thought I'd just as well report all the similar issues I've run into.

### Your environment
* ROS Distro: Noetic
* OS Version: Pop!OS
* Built from source, branch `master` (moveit repo is on commit e7b388d238159de020a9c31f5896112c1513bf6b)

### Steps to reproduce
- Start RViz and moveit
- Load/scan any octomap
- Call service `/clear_octomap`

### Expected behaviour
I would expect the octomap to disappear from the Planning Scene after the service is called.

### Actual behaviour
The octomap is still shown until I reload the MotionPlanning plugin.

atmagopal, 2022-04-29:
I've been experiencing the same issue. Have you had any progress towards fixing this @FSund ?

v4hn, 2022-05-04:
Quickly skimming the code clearing the octomap should trigger `UPDATE_GEOMETRY` which *should* trigger publishing a new ROS message with the cleared octomap. Can you see the message on the `move_group/monitored_planning_scene` topic when you clear the scene?
Either the trigger fails or the new message is interpreted wrongly in the RViz plugin.

FSund, 2022-05-06:
> Can you see the message on the `move_group/monitored_planning_scene` topic when you clear the scene?

Yes, the following message appears on the `monitoried_planning_scene` topic when I press "Clear octomap" in RViz

```
---
name: "(noname)+"
robot_state: 
[...]
world: 
  collision_objects: []
  octomap: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "ab_base"
    origin: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    octomap: 
      header: 
        seq: 0
        stamp: 
          secs: 0
          nsecs:         0
        frame_id: ''
      binary: False
      id: ''
      resolution: 0.0
      data: []
is_diff: True
---
```

The Octomap is still visible in RViz though.

FSund, 2022-05-10:
I am not sure what you are suggesting. Should the `id` field always be filled? Or should it always be filled when the octomap is non-empty (i.e. empty octomap == empty `id` field)?

I'll have a look at the code and try to figure out where everything is being manipulated.

FSund, 2022-05-10:
To me the check [here](https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L1292) seems superflous, since we check `map.data.empty()` and `map.id != OcTree` [here](https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L1368-L1375) anyway. But perhaps I'm not understanding how everything fits together properly.

FSund, 2022-05-11:
It seems the octomap message is created in [PlanningScene::getOctomapMsg](https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L827). The issue I'm having is that the `id` is set in [fullMapToMsg](https://github.com/OctoMap/octomap_msgs/blob/melodic-devel/include/octomap_msgs/conversions.h#L188), so this is not available if `map` does not exist (which I'm guessing it doesn't if the octomap has been cleared).

PR #3134 (2023-03-24): Improve handling of empty octomap messages and "clear octomap" actions
### Description

Currently, if a `PlanningScene` message is published on the `monitoried_planning_scene` topic with an empty `msg.world.octomap.octomap.data` field, the octomap is skipped altogether in if it is a `msg.is_diff` is True ([source](https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L1292)). I think perhaps an empty `data` field is interpreted as the octomap field not being filled. But this makes it impossible to clear the octomap by sending an empty octomap mess [... cut]
closed 2023-03-24 by rhaschke