moveit/moveit #3046: Ask MoveIt to do IK one time only rather than a whole trajectory planning!

mhmiri, 2022-02-01:
Hi,

Q1) I do not want to have trajectory planning of my 6DOF robotic arm. In fact I want to give a start end-effector position and orientation and the finish position and orientation, and have MoveIt to do the IK once and give me the 6 angular rotations of the joins (for example: motor1 needs to rotate 20deg, motor2 -10 deg, ..., motor6 50deg). No need for a whole trajectory of angles.
Is there a way to do this?

Q2) Also sometimes when I use a KDL solver I do not get an answer or the planning, is it because it is randomly choosing methods to find a solution and sometimes it just cannot find it or? or for sure there is no solution? Is their another solver that would do the IK using formulas rather than numerical? I want to use it since I would be doing the IK once (as per Q1) so even if solving it takes a fraction longer it is fine.

Note that this is my first ever question writing on github, so please accept my apologies if Im writing this in a wrong place or format and would appreciate your immediate guidance.

Thanks very much in advance.

rhaschke, 2022-02-01:
This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here only.
The answer to your question depends on how you interface to MoveIt.
- You can directly call API functions as explained in [this tutorial](https://ros-planning.github.io/moveit_tutorials/doc/robot_model_and_robot_state/robot_model_and_robot_state_tutorial.html)
- You can call the `compute_ik` service as explained in [this example](https://github.com/ros-planning/moveit_tutorials/blob/master/doc/kinematics/src/ros_api_tutorial.cpp)

mhmiri, 2022-02-01:
@rhaschke Thank you very much for the links and your guidance.

rhaschke, 2022-02-02:
Please close the issue when your problem is solved.

closed 2022-02-02 by mhmiri