moveit/moveit #3047: Moveit servo PoseTracking object not sending JointTrajectory commands, only sends them when it is stopped

HumbertoE, 2022-02-02:
## Description

A node created in c++ generates a ROS service to be used in python. The ROS service callback calls in a separate thread the moveToPose method from a previously created moveit_servo::PoseTracking object and stops it after 15 seconds. I used as a skeleton the [pose_tracking_example.cpp](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/src/cpp_interface_example/pose_tracking_example.cpp).

For the time being, a publisher is created in python to publish a constant pose at 50 Hz.

## Environment

ROS Distro: Noetic
OS Version: Ubuntu 20.04.3 LTS

## Robot Setup

I am using a real franka emika robot which has a running controller of type position_controllers/JointTrajectoryController. It can be controlled via Moveit (planning and executing trajectories).

## Unexpected Behavior
 
The twist commands are being published at the expected rate (125 Hz), but no commands are being published on the command_out_topic (/position_joint_trajectory_controller/command). At the end, when the moveit_servo::PoseTracking object is stopped with the stopMotion method, 5 messages are being send and the motors of the robot sound as if they activated. I think the stopMotion method sends some las commands to prevent the robot from making a hard stop, but these are the only commands making it to the robot.

The node keeps giving the warning]: "The complete state of the robot is not yet known.  Missing virtual_joint"
and then sends the warning: "Target control rate was missed" once.

## Code and files
The node graph looks as follows:
[image]

and the yaml file for the servoing is:
```yaml
###############################################
# Modify all parameters related to servoing here
###############################################
parameter_ns: "pose_tracking_service"

use_gazebo: false # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.6  # Max linear velocity. Meters per publish_period. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.3 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Rads or Meters per publish period. Only used for joint commands on joint_command_in_topic.
  joint: 0.01
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.

## Properties of outgoing commands
publish_period: 0.008  # 1/Nominal publish rate [seconds]
low_latency_mode: false  # Set this to true to publish as soon as an incoming Twist command is received (publish_period is ignored)

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController or JointGroupPositionController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: trajectory_msgs/JointTrajectory

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

## MoveIt properties
move_group_name:  panda_arm  # Often 'manipulator' or 'arm'
planning_frame: panda_link0  # The MoveIt planning frame. Often 'base_link' or 'world'

## Other frames
ee_frame_name: panda_link8   # The name of the end effector link, used to return the EE pose
robot_link_command_frame:  panda_link0  # commands must be given in the frame of a robot link. Usually either the base or end effector

## Stopping behaviour
incoming_command_timeout:  0.1  # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 4

## Configure handling of singularities and joint limits
lower_singularity_threshold:  17  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 30 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

## Topic names
cartesian_command_in_topic: delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: delta_joint_cmds # Topic for incoming joint angle commands
joint_topic: joint_states
status_topic: status # Publish status to this topic
command_out_topic: /position_joint_trajectory_controller/command # Publish outgoing commands here

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 50 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Two collision check algorithms are available:
# "threshold_distance" begins slowing down when nearer than a specified distance. Good if you want to tune collision thresholds manually.
# "stop_distance" stops if a collision is nearer than the worst-case stopping distance and the distance is decreasing. Requires joint acceleration limits
collision_check_type: threshold_distance
# Parameters for "threshold_distance"-type collision checking
self_collision_proximity_threshold: 0.01 # Start decelerating when a self-collision is this far [m]
scene_collision_proximity_threshold: 0.05 # Start decelerating when a scene collision is this far [m]
# Parameters for "stop_distance"-type collision checking
collision_distance_safety_factor: 1000 # Must be >= 1. A large safety factor is recommended to account for latency
min_allowable_collision_distance: 0.01 # Stop if a collision is closer than this [m]
```

HumbertoE, 2022-02-15:
I solved my issue. It was two mistakes I made when switching the [pose_tracking_example.cpp](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_ros/moveit_servo/src/cpp_interface_example/pose_tracking_example.cpp) to OOP.

1. I  forgot to initialize the value of the node handle attribute nh_ to the node handle given as a parameter. That is:
```cpp
PoseTrackingService(ros::NodeHandle nh): nh_(nh)
  {
  ...
  }
```
This is why my nodes graph looked a bit weird.

2. Before I used ros::spin which didn't allow the method moveToPose of my moveit_servo::PoseTracking attribute to execute properly in a different branch. My work-around was to include the following method and run it from main. Please let me know if you have a better idea for this.
```cpp
  void spinServer()
  {
    // Set up service
    ROS_INFO("Setting up service to move to pose with pose tracker");
    service = nh_.advertiseService("move_to_pose", &PoseTrackingService::move_to_pose_, this);
    ROS_INFO("Service to move to pose with pose tracker ready");
    
    ros::AsyncSpinner spinner(8);
    spinner.start();
    ros::Rate loop_rate(20);
    while(ros::ok())
    {
      loop_rate.sleep();
    }
  }
```

closed 2022-02-15 by HumbertoE

imishani, 2022-02-22:
Hi Humberto! 
Could you please share your script?
Thanks

HumbertoE, 2022-05-17:
Hello Imishani. Sorry for the very late reply. I didn't see your comment before.

Unfortunately, this is only a part of a more complex script we are using and I can't share it. I would offer to do a minimal example, but I am not sure when would I have time for it. If you have more specific doubts, please let me know and I could help with them.