moveit/moveit #3052: Allow removing default collision entries from AllowedCollisionMatrix

simonschmeisser, 2022-02-04:
The API of AllowedCollisionMatrix is somewhat incomplete as it's not possible to remove individual entries from the list of default collisions. Setting the defaultCollision entry for a link to false is however not identical to not having an entry as for this link now all pairs will be checked, including those actually disabled by pair wise entries.

The ros2/MoveIt2 tool changing capability works around this by caching a previous ACM and reverting to it ... which could discard changes made to the ACM elsewhere ... ( https://github.com/PickNikRobotics/tool_changing/blob/main/tool_changing_capability/src/tool_changing_capability.cpp )

Note: I'll likely implement this at some point but feel welcome to beat me to this.

rhaschke, 2022-02-04:
To my understanding pairwise entries always take precedence over default entries (for both, disabling or enabling collision):
https://github.com/ros-planning/moveit/blob/47a66152d5b3559d147b0cf57c4c8bcca50a3931/moveit_core/collision_detection/src/collision_matrix.cpp#L325-L329

However, maybe I misunderstood something. To clarify, could you provide a failing unit test?

simonschmeisser, 2022-02-04:
I have to admit I read the comments in the header instead of the code ... maybe there is a mismatch ... will check again