moveit/moveit #3058: Can I fix demo_gazebo.launch generated by moveit_setup_assistant?

AzulRadio, 2022-02-14:
Environment: Ubuntu 20.04, noetic, moveit: 1.1.8-1focal.20220212.181457 from binary.

When I generate the moveit config package from a .xacro file with moveit_setup_assistant, the generated demo_gazebo.launch is broken. I ran into the same issue as #2210 and #2237 where the robot moves in Rviz but not in Gazebo. With Error msg: 
```
Unable to identify any set of controllers that can actuate the specified joints
```

I looked into this and found that the solution is quite simple (at least for me): append 
```
controller_list:
  - name: <controller_name>
    action_ns: follow_joint_trajectory
    type: FollowJointTrajectory
    default: true
    joints:
      - <joint_name>
```
at the end of `ros_controller.yaml` is sufficient (at least for me) to solve the problem.
(credit:  **venkys** from https://answers.ros.org/question/320990/moveit-unable-to-identify-any-set-of-controllers/)

This bug is too trivial (but cost me 2 weeks to find) and I think maybe I can fix it. But since the 2 issues I mentioned above are all quite old and both closed, I decided to ask before making a PR: **is this a bug already fixed/in fixing? also, is it an appropriate fix?** (well, it seems to be a workaround)

Thanks for your help!

rhaschke, 2022-02-14:
Did you configure controllers in MSA (in the controllers tab)? If you did, there is probably a bug. We welcome any contribution!

AzulRadio, 2022-02-15:
> configure controllers in MSA

@rhaschke Sorry but I have no idea what you are talking about? Here's everything I did to trigger this behavior: 

1. Choose a .xacro file
2. create the moveit_config package with `moveit_setup_assistant setup_assistant.launch`
3. roslaunch `demo_gazebo.launch`
4. in `Rviz`, drag and drop to select the target pos, click `plan & execute`
5. In the terminal you can see the error info.

FSund, 2022-02-22:
@AzulRadio He's probably wondering if you filled in this tab in the MSA (**M**oveit **S**etup **A**ssistant)

[image]

AzulRadio, 2022-03-13:
I test this again with 1.19 again and I think it is already fixed (I believe it was a bug in 1.18). I cannot pin down which commit fixes it or how it is fixed but I am 99% sure this is fixed now. 

**Note**: If you still have the error `Unable to identify any set of controllers that can actuate the specified joints`, please check if you are using `effect_controller` instead of `position_controller`. The `demo_gazebo.launch` seems to only support `position_controller` when I test it.

Thanks to @FSund @rhaschke and anonymous contributors who fixed this!

closed 2022-03-13 by AzulRadio