moveit/moveit #3059: Parameter to limit cartesian speed commanded by pose_tracking.cpp

HumbertoE, 2022-02-15:
## Context

I want to use the PoseTracking::moveToPose method similar to what it's done in the [pose_tracking_example.cpp](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/src/cpp_interface_example/pose_tracking_example.cpp) to go to a pose close (around 10 cm) of the end effector. 

## Description
Now I am getting very high velocities as the desired pose is not as close to the end effector as in teleop applications. This can be dealt with by tuning the PID parameters, but it would be nice to have a parameter to limit the maximum speed sent as a twist command by [pose_tracking.cpp](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_ros/moveit_servo/src/pose_tracking.cpp).

HumbertoE, 2022-02-16:
I made a modification to the [calculateTwistCommand method](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_ros/moveit_servo/src/pose_tracking.cpp#L264-L310), adding two small blocks of code marked as `// Limit angular velocity commands` and  `// Limit angular velocity commands`. I am not sure if this is the best solution, but it is working for me at the moment. I will test it more extensively and reopen the issue if I find some problem.

```cpp
geometry_msgs::TwistStampedConstPtr PoseTrackingLimited::calculateTwistCommand()
{
  // use the shared pool to create a message more efficiently
  auto msg = moveit::util::make_shared_from_pool<geometry_msgs::TwistStamped>();

  // Get twist components from PID controllers
  geometry_msgs::Twist& twist = msg->twist;
  Eigen::Quaterniond q_desired;

  // Scope mutex locking only to operations which require access to target pose.
  {
    std::lock_guard<std::mutex> lock(target_pose_mtx_);
    msg->header.frame_id = target_pose_.header.frame_id;

    // Position
    twist.linear.x = cartesian_position_pids_[0].computeCommand(
        target_pose_.pose.position.x - command_frame_transform_.translation()(0), loop_rate_.expectedCycleTime());
    twist.linear.y = cartesian_position_pids_[1].computeCommand(
        target_pose_.pose.position.y - command_frame_transform_.translation()(1), loop_rate_.expectedCycleTime());
    twist.linear.z = cartesian_position_pids_[2].computeCommand(
        target_pose_.pose.position.z - command_frame_transform_.translation()(2), loop_rate_.expectedCycleTime());

    // Orientation algorithm:
    // - Find the orientation error as a quaternion: q_error = q_desired * q_current ^ -1
    // - Use the angle-axis PID controller to calculate an angular rate
    // - Convert to angular velocity for the TwistStamped message
    q_desired = Eigen::Quaterniond(target_pose_.pose.orientation.w, target_pose_.pose.orientation.x,
                                   target_pose_.pose.orientation.y, target_pose_.pose.orientation.z);
  }

  Eigen::Quaterniond q_current(command_frame_transform_.rotation());
  Eigen::Quaterniond q_error = q_desired * q_current.inverse();

  // Convert axis-angle to angular velocity
  Eigen::AngleAxisd axis_angle(q_error);
  // Cache the angular error, for rotation tolerance checking
  angular_error_ = axis_angle.angle();
  double ang_vel_magnitude =
      cartesian_orientation_pids_[0].computeCommand(*angular_error_, loop_rate_.expectedCycleTime());

  // Limit angular velocity commands
  if(ang_vel_magnitude > omega_max_){
    double factor_omega = omega_max_ / ang_vel_magnitude;
    ang_vel_magnitude = factor_omega * ang_vel_magnitude;
  }

  twist.angular.x = ang_vel_magnitude * axis_angle.axis()[0];
  twist.angular.y = ang_vel_magnitude * axis_angle.axis()[1];
  twist.angular.z = ang_vel_magnitude * axis_angle.axis()[2];

  // Limit linear velocity commands
  double vel_magnitude_2 = twist.linear.x*twist.linear.x + twist.linear.y*twist.linear.y + twist.linear.z*twist.linear.z;
  if(vel_magnitude_2 > (v_max_*v_max_)){
    double factor_v = v_max_ / sqrt(vel_magnitude_2);
    twist.linear.x = factor_v * twist.linear.x;
    twist.linear.y = factor_v * twist.linear.y;
    twist.linear.z = factor_v * twist.linear.z;
  }

  msg->header.stamp = ros::Time::now();

  return msg;
}
```

I also added two parameters to [pose_tracking_settings.yaml](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_ros/moveit_servo/config/pose_tracking_settings.yaml)
```yaml
#####################
# Limiting parameters
#####################
v_max: 0.1
omega_max: 0.1
```

closed 2022-02-16 by HumbertoE

AndyZe, 2022-02-16:
If you would like to open a pull request and have the code reviewed, feel free! Glad you found a nice solution.

Another option might be to use the saturation settings in the PID controllers. I'm not sure they have a saturation option; it's been awhile since I looked at this.