moveit/moveit #3067: moveit_servo does not get planning scene updates from move_group by default

FSund, 2022-02-22:
### Description

`servo_server.cpp` from `moveit_servo` does not provide any arguments to `PlanningSceneMonitor::startSceneMonitor()` ([link to source](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_ros/moveit_servo/src/servo_server.cpp#L66
)). This means that it does not receive planning scene updates given via the service `apply_planning_scene`. This had us scratching our heads for a while.

To receive updates from `move_group` it would need to use the argument `"/move_group/monitored_planning_scene"` (or some portable variant of that).

Is this intentional?

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04

AndyZe, 2022-02-22:
That wasn't intentional - I think nobody has ever noticed or needed this feature before. It looks like the scene monitor defaults to "monitored_planning_scene" at the moment.

Would you suggest putting the planning scene topic in the yaml file, then? Would you like to open a PR for it?

One of my hesitations here is that there are several other topics which take default values, at the moment. Looking at `planning_scene_monitor.h`:

```
  /// The name of the topic used by default for receiving joint states
  static const std::string DEFAULT_JOINT_STATES_TOPIC;  // "/joint_states"

  /// The name of the topic used by default for attached collision objects
  static const std::string DEFAULT_ATTACHED_COLLISION_OBJECT_TOPIC;  // "/attached_collision_object"

  /// The name of the topic used by default for receiving collision objects in the world
  static const std::string DEFAULT_COLLISION_OBJECT_TOPIC;  // "/collision_object"

  /// The name of the topic used by default for receiving geometry information about a planning scene (complete
  /// overwrite of world geometry)
  static const std::string DEFAULT_PLANNING_SCENE_WORLD_TOPIC;  // "/planning_scene_world"

  /// The name of the topic used by default for receiving full planning scenes or planning scene diffs
  static const std::string DEFAULT_PLANNING_SCENE_TOPIC;  // "/planning_scene"

  /// The name of the service used by default for requesting full planning scene state
  static const std::string DEFAULT_PLANNING_SCENE_SERVICE;  // "/get_planning_scene"

  /// The name of the topic used by default for publishing the monitored planning scene (this is without "/" in the
  /// name, so the topic is prefixed by the node name)
  static const std::string MONITORED_PLANNING_SCENE_TOPIC;  // "monitored_planning_scene"
```

AndyZe, 2022-02-22:
It might be easier to just remap certain topics for Servo in your launch files. Sorry for the trouble.

FSund, 2022-02-22:
For our application we depend on octomap collision checking as well (which is disabled in `servo_server`), so we will most likely just implement our own `servo_server`. I think at the very least the current behaviour (and how to enable updates from `move_group`) should be properly documented. What would be the best place to document it?

I have a go at adding the servo topic to the yaml config if you want, that is up to you.

AndyZe, 2022-02-22:
Would you prefer an update to [the tutorial](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html) or a comment in a header file? I'd probably prefer a comment in the header file.

I don't think we should add it the to yaml file because then I'd feel obligated to add all the other PS-related topics.

FSund, 2022-02-23:
I'd vote for both the tutorial and the header. 

I'm not sure what the best way to add it to the tutorials is. Perhaps a new launch-file that includes the topic remapping required to get planning scene updates from move group? Then that file could be referred to in the tutorials.

mrayh001, 2022-08-10:
Hi @FSund  is there any updates to this?

FarStryke21, 2024-10-11:
Hi @FSund 

I think I am facing a similar problem. I am creating a "safety cage" of sorts using collision objects, but the servo server does not seem to detect the collision objects added to the planning scene. Self collision checks work well, but scene collision check does not seem to be working. Any suggestions on how to fix this?