moveit/moveit #3068: "Check if you started movegroups" Error.

ozgursanli7, 2022-02-23:
Hi, I'm currently working on a robotic arm,
Although Rviz is running without any problem in the background, when I try to use joystick control, I get the following error: 
[FATAL] [1645607896.826663]: Check if you started movegroups. Exiting.

-When I launch the demo.launch file, doesn't it mean that movegroups have already started.
-I've tried different setups on different computers as well and I still get the same error. 

ROS Melodic
Ubuntu 18.04

According to the move-it tutorials, It should work without any errors I guess.
This is the exact output I get:

[FATAL] [1645607896.826663]: Check if you started movegroups. Exiting.
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/moveit_ros_visualization/moveit_joy.py", line 42, in <module>
    app = MoveitJoy()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_ros_visualization/moveitjoy_module.py", line 559, in init
    self.updatePlanningGroup(0)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_ros_visualization/moveitjoy_module.py", line 593, in updatePlanningGroup
    raise rospy.ROSInitException(msg)
rospy.exceptions.ROSInitException: Check if you started movegroups. Exiting.
[moveit_joy-2] process has died [pid 5450, exit code 1, cmd /opt/ros/melodic/lib/moveit_ros_visualization/moveit_joy.py name:=moveit_joy log:=/home/umut/.ros/log/f4a8f642-9488-11ec-be3f-7c10c92703fa/moveit_joy-2.log].
log file: /home/umut/.ros/log/f4a8f642-9488-11ec-be3f-7c10c92703fa/moveit_joy-2*.log

adnanhd, 2022-02-25:
Hi, it seems that the **get_group_joint_tips** method of the **RobotInterface** object defined in [moveitjoy_module.py](https://github.com/ros-planning/moveit/blob/3361b2d1b6b2feabc2d3e93c75653f5a00e87fa4/moveit_ros/visualization/src/moveit_ros_visualization/moveitjoy_module.py#L520) at line 525 returns an empty list below
```py
...
class MoveitJoy:
    def parseSRDF(self):
        ri = RobotInterface("/robot_description")
        planning_groups = {}
        for g in ri.get_group_names():
            self.planning_groups_tips[g] = ri.get_group_joint_tips(g)
            planning_groups[g] = [
                "/rviz/moveit/move_marker/goal_" + l
                for l in self.planning_groups_tips[g]
            ]
        for name in planning_groups.keys():
            if len(planning_groups[name]) == 0:
                del planning_groups[name]
            else:
                print(name, planning_groups[name])
        self.planning_groups = planning_groups
        self.planning_groups_keys = (
            planning_groups.keys()
        )  # we'd like to store the 'order'
        self.frame_id = ri.get_planning_frame()
...
```
Since it returns an empty list, following code snipped ([same file](https://github.com/ros-planning/moveit/blob/3361b2d1b6b2feabc2d3e93c75653f5a00e87fa4/moveit_ros/visualization/src/moveit_ros_visualization/moveitjoy_module.py#578) at line 578) raises an IndexError

```py
def updatePlanningGroup(self, next_index):
        if next_index >= len(self.planning_groups_keys):
            self.current_planning_group_index = 0
        elif next_index < 0:
            self.current_planning_group_index = len(self.planning_groups_keys) - 1
        else:
            self.current_planning_group_index = next_index
        next_planning_group = None
        try:
            next_planning_group = self.planning_groups_keys[
                self.current_planning_group_index
            ]
        except IndexError:
            msg = "Check if you started movegroups. Exiting."
            rospy.logfatal(msg)
            raise rospy.ROSInitException(msg)
        rospy.loginfo("Changed planning group to " + next_planning_group)
        self.plan_group_pub.publish(next_planning_group)
```
Therefore you got the following error.
```bash
File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_ros_visualization/moveitjoy_module.py", line 593, in updatePlanningGroup
raise rospy.ROSInitException(msg)
``` 

I don't know moveit/melodic very well. There may be someone who knows possible reasons why **get_group_joint_tips** returns an empty list. As it is written in C++ and wrapped in Python. It is hard to trace **RobotInterface** class. But I assume that it is because you forget something in configurations and setup wizard.

good lucks

rhaschke, 2022-02-25:
Probably you forgot to define an end-effector in SRDF, being attached to a parent link, which becomes the tip link of the parent group.

ozgursanli7, 2022-02-26:
> Probably you forgot to define an end-effector in SRDF, being attached to a parent link, which becomes the tip link of the parent group.

I did actually,

 "
    <end_effector name="arm_gripper" parent_link="link_4" group="metu_rover_arm" /> "
    
This is how it's defined inside the SRDF file.

rhaschke, 2022-03-03:
`get_group_joint_tips()` essentially calls on the group:
https://github.com/ros-planning/moveit/blob/2ab14dee3db52f4a838a9595f22e1c5ac311722f/moveit_core/robot_model/src/joint_model_group.cpp#L500-L525

Could you please debug this further with your specific config? Have a look for the error messages on the console.
You can also try to call the corresponding methods from python, e.g. `RobotInterface.getEndEffectorParentGroup()`.

YeJX0108, 2024-10-04:
> `get_group_joint_tips()` essentially calls on the group: [...]

this works for me ^_^