moveit/moveit #3070: MotionPlanning plugin needs to be manually refreshed in Noetic

mostlyjason, 2022-02-27:
### Description

When I load the MotionPlanning plugin in Rviz, several tabs have boxes which refuse to update or paint. The Joints tab seems to work fine. However, if I select the Scene Objects tab, the contents of the current scene objects does not update. Instead, it displays the contents of the Joints tab. The same thing happens for several other tabs. This only happens on Noetic. When I build and run Melodic it works fine. 

I can also force it to paint the contents of the screen by manually removing the reloading the plugin.

### Your environment
* ROS Distro: [Kinetic|Melodic|Noetic]
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

simonschmeisser, 2022-02-27:
I haven't seen that myself. Please fill out the template when reporting an issue, especially the "your environment" part. Please also add if you're running wayland or xorg and the output from starting rviz

mostlyjason, 2022-02-27:
Hang on, I submitted this accidentally,will reopen when I have the rest of the details

mostlyjason, 2022-02-27:
Ok I added more details in my comment above. I filled out the environment section and I'm using xorg.

simonschmeisser, 2022-02-27:
does resizing the pane help?

mostlyjason, 2022-02-28:
Resizing looks like it triggers repaint, but there is something wrong with the buffer. The frame looks fine, its just the panel contents and header that look wrong. 

[image]

mostlyjason, 2022-02-28:
Interestingly, the mouse controls for the joint positions work fine, even though they are not being rendered correctly. Also the rest of the Rviz window and other plugins look fine. Only the MotionPlanning one is broken, and only specific tabs.

v4hn, 2022-02-28:
This is probably related to your docker setup and not to MoveIt, though the error is apparently very consistent...
Did you try to reproduce this with [x11docker](https://github.com/mviereck/x11docker)?

Maybe @rhaschke has seen this before?

rhaschke, 2022-03-03:
This rendering issue is probably related to Qt and/or your docker config. I'm pretty sure it's _not_ MoveIt related. We don't do special (paint) event handling...

rhaschke, 2022-03-04:
Closing, because unrelated to MoveIt.

closed 2022-03-04 by rhaschke