moveit/moveit #3071: Unable to to connect to move_group action server when creating a MoveGroupCommander from remote node via python interface

adamlouis, 2022-02-27:
### Description

I have 2 computers.

On one computer (computer A), I am running ROS core + MoveIt server.

On the other computer (computer B), I am running python code using the MoveIt python interface ("controller code").

If I run the controller code on computer A, everything works as expected.

If I run the controller code on computer B, I repeatedly get the error below while creating a `MoveGroupCommander`, even though creation of `RobotCommander` and `PlanningSceneInterface` succeed. The error is:

`RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)`

This is the failing call [in the C++ code](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L155).

My suspicion is that some is wrong here with namespaces. I've seen some potentially-relevant issues, PRs, & threads like:

- https://github.com/ros-planning/moveit/pull/2922
- https://github.com/ros-planning/moveit/issues/2040
- https://github.com/ros-planning/moveit/issues/1807
- https://github.com/ros-planning/moveit/issues/1576
- https://answers.ros.org/question/361726/unable-to-connect-to-move_group-action-server-move_group-within-allotted-time/

I am relatively new to ROS and MoveIt so not sure if I am missing something obvious here.

Would appreciate any help or direction.

Thanks for all of your work on MoveIt!

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? Binary
* Binary: 1.1.8

### Steps to reproduce

1. On computer A, run a roscore + MoveIt server. In my case, I am using code from interbotix that is similar to the MoveIt tutorial:

`roslaunch interbotix_xsarm_moveit xsarm_moveit.launch robot_model:=px150 use_fake:=true dof:=5` ([link to github](https://github.com/Interbotix/interbotix_ros_manipulators/tree/main/interbotix_ros_xsarms/interbotix_xsarm_moveit), [link to docs](https://www.trossenrobotics.com/docs/interbotix_xsarms/ros_packages/moveit_motion_planning_configuration.html))

2. Copy this code onto computer A and computer B to a file `main.py`:

```
import sys
import rospy
import moveit_commander

# you can change these if you want to test with some other robot
robot_name = 'px150' # you could also read this from robot.get_group_names()[0] ... it works!
node_name = 'moveit_test'
group_name = 'interbotix_arm'

moveit_commander.roscpp_initialize(sys.argv)
rospy.init_node(f'{node_name}', anonymous=True, log_level=rospy.DEBUG)

# this works on computer A and B
robot = moveit_commander.RobotCommander(
    ns=f'/{robot_name}', 
    robot_description=f'/{robot_name}/robot_description',
)

# this works on computer A and B
scene = moveit_commander.PlanningSceneInterface(
    ns=f'/{robot_name}', 
)

# this works on computer A BUT **NOT** on computer B
commander = moveit_commander.MoveGroupCommander(
    f'{group_name}',
    ns=f'/{robot_name}', 
    robot_description=f'/{robot_name}/robot_description',
    wait_for_servers=5,
)

print('SUCCESS!')
```

3. On computer A, run the code specifying localhost env vars:

`ROS_MASTER_URI=http://localhost:11311 ROS_IP=127.0.0.1 ROS_NAMESPACE='' python3 main.py`

4. Observe that computer A prints `SUCCESS!`

5. On computer B, run the code specifying remote host env vars:

`ROS_MASTER_URI=http://<your remote IP>:11311 ROS_IP=<your remote IP> ROS_NAMESPACE='' python3 main.py`

6. Observe that computer B creates the `robot` and `scene` objects but fails while creating the `commander` object with:

`RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)`

### Expected behavior

Computer B should create a `moveit_commander.MoveGroupCommander` and print `SUCCESS!`

### Actual behavior

Computer B fails with error:

`RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)`

while creating the `MoveGroupCommander` after succeeding on `RobotCommander` and `PlanningSceneInterface`

### Backtrace or Console output

```
$ ROS_MASTER_URI=http://<REMOTE IP>:11311 ROS_IP=<REMOTE IP> ROS_NAMESPACE='' python3 main.py
[DEBUG] [1645989676.847731]: init_node, name[/moveit_test_67499_1645989676629], pid[67499]
[DEBUG] [1645989676.855024]: binding to 0.0.0.0 0
[DEBUG] [1645989676.861092]: bound to 0.0.0.0 42073
[DEBUG] [1645989676.867550]: ... service URL is rosrpc://<REMOTE IP>:42073
[DEBUG] [1645989676.875051]: [/moveit_test_67499_1645989676629/get_loggers]: new Service instance
[DEBUG] [1645989676.885703]: ... service URL is rosrpc://<REMOTE IP>:42073
[DEBUG] [1645989676.891342]: [/moveit_test_67499_1645989676629/set_logger_level]: new Service instance
[ INFO] [1645989676.948158097]: Loading robot model 'px150'...
[ INFO] [1645989676.956551110]: No root/virtual joint specified in SRDF. Assuming fixed joint
Traceback (most recent call last):
  File "move.py", line 22, in <module>
    commander = moveit_commander.MoveGroupCommander(
  File "/opt/ros/noetic/lib/python3/dist-packages/moveit_commander/move_group.py", line 66, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(
RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)
```

v4hn, 2022-02-28:
If setup correctly ROS is network transparent so if it works locally but not remotely you either have different launch files or your network setup is wrong. Try to take moveit out of the equation and just run `rostopic pub /foo std_msgs/String "data: 'foo'"` on one machine and `rostopic echo /foo` on the other. Then try the other way. I suspect at least one direction will fail.

You might have to set your server to it's network IP/name instead of using `localhost`.
I would also advise against using `ROS_IP` and instead use `ROS_HOSTNAME` with properly setup DNS resolution, but that might just be because of issues I experienced years ago.

Anyway, unless you can guarantee this is MoveIt specific, I'll close it here.

closed 2022-02-28 by v4hn

adamlouis, 2022-02-28:
Thank you @v4hn!

Your small `/foo` example demonstrated the issue. I got it working now.

I made 2 poor assumptions:

- I assumed that because my  `RobotCommander` and `robot.get_group_names()[0]` calls worked, my ROS node network communications were configured correctly. 
- I had assumed equivalence of IP vs. hostname within `ROS_MASTER_URI` 

To resolve, I did the following:
- `unset` ROS env vars related to networking on both computers to support a clean experiment (`ROS_MASTER_URI`, `ROS_HOSTNAME`, `ROS_IP`, `ROS_NAMESPACE`)
-  update the `/etc/hosts` files with hostnames & IPs on each node to ensure proper DNS resolution
- on computer A:
    - `roscore`
    - `rostopic echo /foo`
- on computer B:
    - `ROS_MASTER_URI=http://<remote hostname>:1311 rostopic pub /foo ststd_msgs/String "data: 'foo'"`

Once this small experiment worked, the code I posted above worked.

In the end, my issue was trivial & not related to MoveIt (as you said), but my 2 bad assumptions led me astray.

Thanks again for your help!