moveit/moveit #3074: Pilz Industrial Motion and trajectory interpolation, how increase the number of generated points?

Muhrobot, 2022-03-02:
Hello!

I'm using in a project the planner Pilz Industrial Motion, ROS noetic and Moveit 1. I can plan and execute PTP, LIN and CIRC moves, and get the points with the generated trajectory. I know PILZ plans the trajectory and divides the segments by time, each 0.1 secs has a new joint goal, sometimes it makes enough points but, other times, i need a denser trajectory ( Maybe dividing each 0.5 secs or other values less than 0.1 ). 

After these analyses, i start to search in the lib how it's divided. So i find this:
http://docs.ros.org/en/melodic/api/pilz_trajectory_generation/html/pilz_trajectory_generation/trajectory__generator_8h_source.html 

In the 91 line, I see a variable named sampling_time, and the value is 0.1. That's my key! This var. appears in others archives and all related with generating or planning the trajectory. But it is not so simple, this archive in docs.ros belongs to Moveit Source Build, and I'm using a  Binary install. I can easily find and change this var in Source install, and I get nothing in binary.  I don't want to change my moveit build, and I question myself if there is a way to change it or if anyone already does that. I  search in my folders, even open .deb and .so archives with the software GHEX( that translates from binary to english ) , but i can't find this var.

So, how do I find this var or, change how PILZ makes the interpolation of the points?
I Hope someone can help me!

v4hn, 2022-03-03:
Hi @Muhrobot ,

thanks for reporting this shortcoming. I just changed the title to match the issue.
The constant in MoveIt is hardcoded [here](https://github.com/ros-planning/moveit/blob/master/moveit_planners/pilz_industrial_motion_planner/include/pilz_industrial_motion_planner/trajectory_generator.h#L110) in the base class of the trajectory generators as a default parameter.
The method is invoked from the base planning context [here](https://github.com/ros-planning/moveit/blob/master/moveit_planners/pilz_industrial_motion_planner/include/pilz_industrial_motion_planner/planning_context_base.h#L131) but that does not set the optional argument.

A more appropriate solution could turn `sampling_time` into a planner parameter that can be set in the moveit configs.
That still would not allow to change it easily at runtime, but we could at least configure it.

The best workaround for this right now would be an additional post-processing filter in the planning pipeline to resample at your preferred resolution. [ros-industrial has a plugin for just this purpose](https://github.com/ros-industrial/industrial_core/blob/melodic-devel/industrial_trajectory_filters/planning_request_adapters_plugin_description.xml#L13-L22).

Muhrobot, 2022-04-07:
> Hi @Muhrobot ,
>
> thanks for reporting this shortcoming. I just changed the title to match the issue. The constant in MoveIt is hardcoded [...]

Thanks for the answer @v4hn. I try to find sampling_time in the moveit binary version, but i didn't get it. It's seems possible in the source version, but not in binary.
I will follow your idea about post-processing, that a good way to configure the number of points. Thanks!

v4hn, 2022-04-08:
even if a weird workaround exists, it should be configurable in the planner as I outlined above.

Muhrobot, 2022-04-11:
> even if a weird workaround exists, it should be configurable in the planner as I outlined above.

Hello @v4hn !